HiddenRamblings/TagMo #101: Initialize TP-data

Bombastisch, 2017-11-26:
Can we initialize TP-data?

I tried to make it work myself and did a bit of investigation, but my lack of programmin knowledge brought me to an end.

I'd like to show you my ideas and maybe somebody @North101 @HiddenRambler can implement it.

First this was my idea to make a "Formatting" option: 

[image]

Which then calls the formatting function.

[image]

According to my investigation these bytes are the only ones that need to be changed.

- grey: The APP-ID must be set to the APP-ID from Zelda: TP (HD)
- blue: only the underlined 4 bytes (1. byte: 0xFF or 0x00, 2. byte: shadow cave level: [maxed: 0x28], 3. byte: wolf-link's hearts [maxed: 0x50], 4. byte always 0x64)

The areas around it (1 line above & 2 lines below this "important bytes" are initialized to 0x00, but I am not sure if it is necessary to initialize.

I really hope somebody can do a better job than me and finally fully implement TP support.

Thanks in advance! ;)

North101, 2017-11-26:
I've had a look at it before but I couldn't get it to work. I might have been doing something wrong or there might have been some extra data required to be set

North101, 2017-11-27:
Just tested. I set all the hex data on a blank Wolf Link and it didn't work, so there must be something else needed. Also tried changing the data on a 20 heart Wolf Link to the above and it still worked.

Would be interesting to get dumps of the same Wolf Link at various stages.

Bombastisch, 2017-11-27:
Did you test it in TP or Botw? Because maybe TP needs the Amiibo registered  in it's files at first use. And so TP didn't  know what to do with it.  (So its a Game based problem and not Tagmo's) Do you own Botw? Maybe you should try it there, because Botw doesn't have the "shadow cave leveling system" but just spawns wolf link in game.

North101, 2017-11-27:
I tested in BotW Switch version. Setting a "blank" Wolf Link to Cave level 40 and 20 hearts spawned the default 3 heart Wolf Link while setting that same data on a Wolf Link that had cleared Cave of Shadows in TP worked fine and I could adjust the amount of hearts.

North101, 2017-11-28:
Figured it out. You also need to set the timestamp/count (hex: 0x28 till 0x33)

Bombastisch, 2017-11-28:
That's really awesome! Looking forward for a PR in future time. :)

North101, 2017-11-28:
>future time. :)
#103 :)

Bombastisch, 2017-12-19:
I think since you implemented it nowwe can close this.  Awesome ;)

closed 2017-12-19 by Bombastisch

pythonimus, 2018-08-31:
Will there be a realease with this PR added or is the project abandoned now?

Bombastisch, 2018-08-31:
The project is not "abandoned" but the two maintainers run this project for free so it can be that a project is progressing slowly. @North101 stated somewhere else that he has a lot to do irl and this means that we maybe even have to wait longer.

Meanwhile you can download the sourcecode from https://github.com/HiddenRamblings/TagMo/tree/experimental_ui and compile it in Android Studio! It's very easy even for a not programmer.

According to North101 it's not 100% complete yet, but it comes with already lots of improvements that will also be necessary for the upcomming Smash Bros Ultimate!

I hope the devs are having a good time and will return soon! :)