HiddenRamblings/TagMo #102: Can't load key files.

Crackst, 2017-11-27:
Hello,
I'm using a Samsung Galaxy S5 and whenever I try to load in the keys I get a message saying "Error: uri". I tried different releases, it's the same issue on every version tho. I also tried different Keys, but again, same issue.
Screenshot shows the messages I'm getting on 2.3.2 and 2.6.1 (on 2.3.2 the error message pops up, on 2.6.1 theres only the grey bar on the bottom) 
[image]
[image]

closed 2017-11-27 by Crackst

Bombastisch, 2017-11-27:
Could you replicate your problem again, and then click on the 3 dots in the mainmenu and "dumb logcat" and please send the logcat.txt in this thread. Then our devs will be able to fix it

Bombastisch, 2017-11-27:
Closed? Did it work now?

Crackst, 2017-11-27:
It did work. I just had to use another file manager, the Samsung one caused the Problem. Thank you for your reply anyways!