HiddenRamblings/TagMo #104: Improving amiibo editing

North101, 2017-11-28:
Right now we can only edit TP and SSB Amiibo data. We could instead allow editing all Amiibo data (that make sense) and have a "plugin" system for editing app data based on the app id.

Editable data:
- Country code
- counter
- init date
- modified date
- Amiibo nickname
- Mii owner
- app id
- app data
- NTAG UID?
- Amiibo ID?

Bombastisch, 2017-11-28:
Awesome!

NTAG UID is locked by manufacturer, this option only works with powertags.

Amiibo ID: I think that bytes are locked as well.

North101, 2017-11-28:
When writing to an already written tag, they are locked. But to a new tag they wouldn't be (unless its linked to the crypto stuff)

North101, 2017-12-10:
[image]
[image]
[image]

North101, 2017-12-10:
https://mega.nz/#!ZyRjSI5R!q4e9Iy1fjU7hnnRxdvRertnxsJMdRLnXQnYYWvEtXPY

Bombastisch, 2017-12-30:
Sorry for responding so much later, but I did some testing and found some bugs etc.

First things first, I love the new layout! :D 

**Questions:** 
- What does "Ignore tag restrictions" mean?
- Do you want to drop QR-code support?

**Maybe Bugs?**
- Not all region codes are detected. e.g German region code always says undefined in Tagmo.
- Write counter is not incremented when setting  Appdatas (e.g initialize Wolf Link)
- The reset dump/bin for PowerTags cannot be written, window doesn't open.
- When NFC is not enabled a false warning text shows up, when trying to write

**Visual stuff**
- Setting the menu view to "Large" adds a blue bar underneath all BINs in the list. Looks confusing and unnecessary
- No more warnings to load keys when starting the app for the first time
- Why is the "edit tag data" button hidden in this "3 dot menu", I would move it next to the Restore button

Thank you for updating the old UI, I love it and I hope you will continue to work on it :)

ManCaveMedia, 2020-08-10:
I'm sorry to revive an old issue. Unfortunately I don't have the knowledge and experience needed to merge the recent  commits from master2 and build/compile my own apk.

Could someone build an updated version of this experimental issue? I've used it in the past and I remember I really liked it! Would appreciate it.

AbandonedCart, 2021-09-11:
This has now become part of the released experimental build. Thanks @North101 for all of the UI improvements.

closed 2021-09-11 by AbandonedCart