HiddenRamblings/TagMo #105: 'Unknown' character name when scanning Amiibo

commorancy, 2017-11-30:
I'm brand new to all of this. I'm using 2.6.1. If I'm missing something, please guide me. When I scan both the _Urbosa_ and _Daruk_ actual Amiibos, they come up with all of the correct data about the Amiibo including the image, but the name is red and says 'Unknown'. Is there a way to edit the Amiibo's name to correct this problem?

[image]

hwchoy, 2017-12-01:
Did you go into the settings and "Sync Amiibo info with AmiiboAPI"?

The champion names show up fine for me.

commorancy, 2017-12-01:
No, I hadn't tried that. However, after syncing with the API, it seems to have solved the problem. Thanks for the suggestion.

closed 2017-12-01 by commorancy

Bombastisch, 2017-12-02:
@commorancy The thing is Nintendo adds Amiibos more frequently as we release a new version of TagMo, so there might be some Amiibos showing up unknown. To solve this problem a very friendly dev introduced syncing the Amiibo data from a webservice, this allows you to always have all the Amiibos shown up in your list, without needing to redownload a new update of the complete app.