HiddenRamblings/TagMo #107: Tag Was Lost

baifag, 2017-12-14:
Hi, having problems scanning any 215 tag with the app, constantly either getting tag was lost or tag lost exception. I've tried old and new version of the app, using an Honor 6x which has NFC but just seems to have problems with the app.

rehv, 2017-12-15:
I get it everytime too with my Moto Z2 Play. It says Tag Lost but then if I keep it there it'll scan just fine. I'm not sure if it is an app problem or sensor problem because it's the first time I'm using it on this phone.

baifag, 2017-12-15:
How long would you say it takes to write/scan in tagmo? Other apps read the cards just fine but using Tagmo just instantly gives this error, ive tried leaving it unmoving but it just seems to be doing nothing till i wiggle it about which gives rapid-fire errors, maybe im a retard and need to give it more time.

rehv, 2017-12-15:
Well, do you know where exactly is your NFC sensor located on your phone? Mine only works if I put the tag near the frontal flash. It takes 2 seconds to scan the tag... It's like 1 second to scan the tag and get TAG LOST, then another second to scan again and load the amiibo correctly.

baifag, 2017-12-15:
Yeah around the same place, I just tried it again barely moving it from the sensor and nothing just rapid fire rejections, more specifically;
Error while writing main data (stage 1) java.IOException: Tranceive failed
Error while writing main data (stage 1)android.nfc.TagLostException: Tag was lost.

baifag, 2017-12-15:
I've tried what feels like almost all on the releases page as well as a few dotted around that are either old release or custom ones for other people with issues, all have had the same error. Thios time I tried the newest one 2.6.1.

rehv, 2017-12-15:
Okay, I've done further testing and it appears I'm not able to write to the tag due to the first "tag lost". The situation I described only happened while reading them. This is the first time I'm testing this apk on this phone, but IIRC 2.3. something worked fine for reading/writing with my old Moto X2 phone.

baifag, 2017-12-15:
Hmmm, thanks! I'll double check and go through them all and give them a try.

rehv, 2017-12-15:
I don't know exactly what I did but I was just able to write a SSB Robin tag with the 2.6 app. I think it's better if we just wait if the author can figure something out.

rehv, 2017-12-15:
Okay, I figured it out... It's the same as reading, but writing takes a little longer to "retry" or something. I kept the tag near the sensor and waited until it said it was done. Just wrote a SSB Zero Suit Samus on the first try.

baifag, 2017-12-15:
Ok so trying 2.3.4 beta it seem to write one with no error appearing, however it isnt loading it into the game so im not sure if writing was actually complete or not, there was no confirmation in the app just the amiibo i had selected was not loaded anymore, still unable to just scan them annoyingly.

rehv, 2017-12-15:
Try to scan the tag using the app again, see if is still blank (it probably is).

baifag, 2017-12-15:
That's what I mean, using the Scan Tag feature on the app still gives the error I always get

rehv, 2017-12-15:
Okay, try this. When I approach a blank tag without selecting any function in TagMo app, it just says "New tag collected - empty tag" with no errors whatsoever. This way you'll be able to check if they're blank at least.

baifag, 2017-12-15:
Aye bang on, New tag collected Tag is empty or phone is not aligned. So Writing isn't giving me an error anymore but it also isn't writing.

rehv, 2017-12-15:
Would it help if I tried to make a video showing what I do to scan the tags?

baifag, 2017-12-15:
Unless I actually have Alzheimer's when trying to put the card to the phone I don't think its that, I honestly just assumed it was a problem with the app and my phone, im loading the app, applying the unfixed key and locked key, clicking load tag, selecting an amiibo, selecting write tag (auto) then holding my card to the nfc reader right?

rehv, 2017-12-15:
Yes, pretty much that. The keys file only needed to be loaded once, and the V2.6+ app only has write tag on default screen (which is the same as Write Tag (auto) on earlier versions).

Can you try looking at this link? https://gbatemp.net/threads/release-tagmo-tool-for-creating-editing-amiibos.431670/page-7#post-7170568

A Honor Phone user says the latest version [here](https://github.com/azsde/TagMo/releases) is the only one that works for his/her phone.

baifag, 2017-12-15:
I'm just constantly re-applying the keys becuase every new attempt is with a new install and version. Will try out what you've linked but the download isn't a .apk?

rehv, 2017-12-15:
Oh, wait, I'll try to get the .apk for you. Didn't try downloading it before sending to you.

Bombastisch, 2017-12-15:
Ohh, what a nice thing to see helping each other out. Love it :) ! @baifag , could you please uese the latest version of TagMo again and after the writing error happened go to the main menu and click on the three withe dots in the upper right. Then click "Dump Logcat" and send the .txt file into this conversation. Maybe the devs can help you out then and fix the problem :)

baifag, 2017-12-15:
[tagmo_logcat[1].txt](https://github.com/HiddenRamblings/TagMo/files/1563272/tagmo_logcat.1.txt)

Here's the logcat from the most recent Tagmo release.

Bombastisch, 2017-12-17:
@HiddenRambler @North101 can you dig into this?

HiddenRambler, 2018-01-30:
@baifag  can you do a read of the tag you wrote and post the file and the logcat for me.

Bombastisch, 2018-03-14:
@baifag Is there any news on this or does it still not work?

Bombastisch, 2018-06-05:
Closing, since it appears that there is no more feedback. If the problem still occurs feel free to open a new issue thread. :)

closed 2018-06-05 by Bombastisch