HiddenRamblings/TagMo #108: Source UID does not match the target

marcored331, 2017-12-16:
hi when i try to write on nfc a new tag, with write it say that the tag was already an amiibo and i must use restore tag, when i use restore tag appear "Source UID does not match the target". How i can fix this error ?
thansk

Bombastisch, 2017-12-16:
Are you sure it is a "new / blank tag" or if it is already a amiibo?

marcored331, 2017-12-18:
it's already an amiibo

Bombastisch, 2017-12-19:
You can't overwrite a Amiibo with new Amiibo data. Common mistake. You need to use blank tags to make new Amiibos

closed 2017-12-19 by Bombastisch

LucianoMP, 2019-10-06:
I know I'm a bit late but... what's the restore button for?

Bombastisch, 2019-10-06:
For example:
If you edit Wolf Link's hearts in TagMo you need to write back that custom data to the amiibo. You need to use the Restore Button for this action.
You don't write or _create_ a new amiibo, but restore edited data to a existing amiibo.

When restoring data to amiibo the universal ID (UID) of the loaded Bin must match the NTag you are restoring to. This is simply done automatically just by scanning the real amiibo -> editing the data -> restoring the new data.