HiddenRamblings/TagMo #109: Failed to decrypt tag

Postella, 2017-12-18:
Hey,

I use the latest version 2.6.1. and a galaxy S7, nfc 215. For some unknown reasons, I can't  load any other tags than the zelda tags. I tried to load the golden mario amiibo, peach amiibo or a single smash bros amiibo .bin files and every time it says "failed to decrypt tag" and the app says everything (game series and so on) is unknown.
I thought that was because the bin file from golden mario was uploaded in 2015, but then I downloaded a mario bin file, that was uploaded in November 2017 and it doesn't work either.

Anyone know how to fix it?
Thanks.

Bombastisch, 2017-12-19:
That's an unfamiliar problem...    So you problem is loading the tags from a .bin file and not scanning them, right?

Maybe you should reinstall the app and try again. Let me know what happend.

Postella, 2017-12-19:
Hey,

I think I've solved it, lol. I've downloaded the bin files via my computer and transferred them to my mobile phone and it has worked! Now TagMo shows the picture, game series and so on and even the option "Write Tag" works perfectly. The file sizes of the files I downloaded via my mobile phone were around 35kB (these were the files that didn't work), but the correct bin files were just 540 bytes. 
I don't know why downloading via phone didn't work and why the file sizes are so big. So that's it, I tried the peach amiibo and the other mario amiibos, they worked perfectly.

Bombastisch, 2017-12-19:
Yes that seemed like a bad file, you downloaded on your phone. Amiibo dumps are always only 540 bytes small.

Postella, 2017-12-19:
I will close this topic but first let me clear: The amiibo file (e. g. Mario) I've downloaded via phone (with 35 kb) was exactly the same(!) from the same source as the bin file from Mario I've downloaded via my computer (but in this case with 540 bytes). I don't know why the same file has different file sizes relative to the way of download. It's mysterious^^

closed 2017-12-19 by Postella