HiddenRamblings/TagMo #111: Cant find keys

Ndog28, 2017-12-20:
I dont know if ive downloaded the wrong keys but its saying the key signature doesnt match im using an Lg V30 if that helps

Bombastisch, 2017-12-21:
You probably don't have the right keys. Try to get them from another source

Ndog28, 2017-12-21:
This is the only website for the keys right?

Bombastisch, 2017-12-21:
The keys aren't availible on Github. You need to find them yourself somewhere else.

Ndog28, 2017-12-25:
> Can I close this issue?

I still havent solved it yet been pretty busy lately lol

Bombastisch, 2017-12-25:
The key file/s need to be obtained from somewhere else. We can't share them due to copyright issues. Your fave search engine might help^^

Ndog28, 2017-12-25:
Oh i get it now lol i got a new phone and i forgot how to find the keys
again thanks for the help

closed 2017-12-27 by Bombastisch