HiddenRamblings/TagMo #117: Error while writing main data (stage 1) java.io.IOExeption: transience failed

Swaggertron182, 2018-02-03:
Trying to write amiibo to NTAG215 cards but comes up with this error, use Samsung galaxy s7 edge.

Bombastisch, 2018-02-03:
Go to the settings and disable "Tag Verification" and retry. Please send me your results

HiddenRambler, 2018-02-07:
Did you buy the same card as #120 ?

Swaggertron182, 2018-02-08:
Thank you for the help, I'll try tonight, keep on hacking amiibos, PS I'm an android dev

Bombastisch, 2018-02-12:
Any news on this? :)

charlio85, 2018-02-12:
Hi. I'm having the same problem with my Samsung Galaxy S5. I've tried different versions (2.6, 2.5, 2.4...) and tried disabling "Tag Verification" but I'm always having the same problem with scanned or downloaded tags.
I verified that my cards are NTAG 215 too but here's the scan result:
[image]
[image]
[image]
[image]
[image]

I've tried with several cards from the package I bought and the result is always the same. I hope it helps... but I'll also apretiate any help to me :)

Bombastisch, 2018-02-12:
Hi, yes these seem like the Tag in #120 . You should return them and order working ones. I can send you a link which I bought, but I can't guarantee that the seller didn't change its product like the one in #120

charlio85, 2018-02-12:
Yeah, certainly my cards should be the same as in [#120](https://github.com/HiddenRamblings/TagMo/issues/120). I'm getting a refund from Amazon and taking other ones, so I would appretiate that link. At least a possitive verified experience will help to make the choice easier.

Bombastisch, 2018-02-12:
https://www.amazon.de/Sticker-Aufkleber-amiibo-Figuren-Stück-kompatibel-Weiß/dp/B06XZR4JM9/ref=redir_mobile_desktop?_encoding=UTF8&__mk_de_DE=ÅMÅZÕÑ&dpID=31UFZfY%2BYAL&dpPl=1&keywords=amazon%20ntag215&pi=AC_SX236_SY340_FMwebp_QL65&qid=1512752657  These are the ones I bought. Thy work perfectly with my phone. Can't guarantee that these Tags work still tho.

charlio85, 2018-02-18:
Hi. Finally I've purchased these ones:
[https://es.aliexpress.com/item/Free-Shipping-50pcs-Lot-NTAG215-NFC-Cards-NFC-Forum-Type-2-Tag-13-56MHz-ISO-IEC/32802097835.html?spm=a2g0s.9042311.0.0.EXiMBA](https://es.aliexpress.com/item/Free-Shipping-50pcs-Lot-NTAG215-NFC-Cards-NFC-Forum-Type-2-Tag-13-56MHz-ISO-IEC/32802097835.html?spm=a2g0s.9042311.0.0.EXiMBA).
In the recent comments people say they they work perfectly for amiibo too. They should arrive in a few less than a couple of weeks and once I try them I'll let you know if they work.

Bombastisch, 2018-03-14:
Any news on this @charlio85 ? :)

closed 2018-06-04 by Bombastisch