HiddenRamblings/TagMo #118: Error while writing main data (stage 1) java.io.IOException: Transceive failed

HezzyUK, 2018-02-03:
Hello,

Please see attached log and scan of the card I'm attempting to use. 

I'm on a Samsung S8+ running Android 7.0

[TagInfo scan_ NXP Semiconductors NTAG215 tag.txt](https://github.com/HiddenRamblings/TagMo/files/1691904/TagInfo.scan_.NXP.Semiconductors.NTAG215.tag.txt)
[tagmo_logcat.txt](https://github.com/HiddenRamblings/TagMo/files/1691905/tagmo_logcat.txt)

HiddenRambler, 2018-02-07:
Did you buy the same card as #120 ?

HiddenRambler, 2018-02-09:
Please see my last comment on #121

HezzyUK, 2018-02-09:
Yes, I did buy the same cards as #120

Bombastisch, 2018-02-09:
So we know that those are bad tags, which got "pre-locked".   Maybe we should add some feature which test the Tags before writing to then output an error message that the tags are prelocked so users can use them. Can this Issue be closed?

HezzyUK, 2018-02-09:
I have re-ordered some new tags and they arrive tomorrow. If all is well, I will close then?

closed 2018-02-09 by Bombastisch