HiddenRamblings/TagMo #119: The phone can be used directly as a nfc label, switch directly read amiibo from the phone??

lybest, 2018-02-05:
Mobile nfc can be used directly as a bus card, e-bank card payment, why can not directly use the phone as amiibo, app should have a way to achieve it

Bombastisch, 2018-02-05:
> CyanogenMod had HCE before Android did, in CM 9. It was such a big thing that Android added HCE in the next release (KitKat, I believe). But they didn't base it on CyanogenMod code, instead they used a different library and different API that is stricter in terms of what cards you can emulate. CyanogenMod switched to the AOSP implementation after CM 10.2, and so no current version of Android or any of its derivatives is capable of emulating an Amiibo. Enabling this behavior would require something between a custom ROM and an Xposed module (or would not be at all possible without root and an unlocked bootloader). Even then, it will be very difficult.

https://www.reddit.com/r/amiibros/comments/4omesh/send_amiibo_directly_from_android_to_3ds/

lybest, 2018-02-06:
Oh I see

closed 2018-02-06 by Bombastisch

FunDeveloping, 2019-06-03:
I have CM9 with an NFC compatible phone, Any way to go by doing it?