HiddenRamblings/TagMo #120: Error while writing main data. Transceive failed. S8+

CheeseJuggernaut, 2018-02-06:
I'm using version 2.6.1 of Tagmo with my S8+ and these NFC cards: http://amzn.eu/7N4BkRd.

I've tried disabling wifi and other similar settings, scanning the card in different positions, and version 2.5.1 of Tagmo.

Attached is my logcat 
[tagmo_logcat[1].txt](https://github.com/HiddenRamblings/TagMo/files/1697567/tagmo_logcat.1.txt)

CheeseJuggernaut, 2018-02-06:
Just tried with an S7 Edge, same problem

Bombastisch, 2018-02-06:
Samsung phones seem to have a problem. I think we need a dev to finally fix the problem with Galaxy phones. Did you try to disable tag verification in the settings and write then? I think it's worth trying. Please post your results :)

networkfusion, 2018-02-06:
My S8+ can write to my (https://zipnfc.com/nfc-cards-and-discs/nfc-tag-pvc-card-credit-card-size-inkjet-ntag215.html) nfc cards fine.

CheeseJuggernaut, 2018-02-06:
I've tried disabling the tag verification. It just makes it take an extra second before the error pops up.

CheeseJuggernaut, 2018-02-07:
@networkfusion What version of Tagmo do you use? Whats your setup?

CheeseJuggernaut, 2018-02-07:
I just tested with a Galaxy S5 and had the same results. I doubt Samsung phones 4 generations apart would have the same issue, so I'm gonna guess that its the NFC tags. I'm ordering some new ones, but I am also wondering what could be causing these tags to fail

CheeseJuggernaut, 2018-02-07:
Here I scanned my NFC tag, but I cant seem to upload the original xml format file. Here is a txt version, hope it helps.
[taginfo_scan[1].txt](https://github.com/HiddenRamblings/TagMo/files/1702673/taginfo_scan.1.txt)

Bombastisch, 2018-02-07:
@CheeseJuggernaut samsung Galaxy phones always have problems to write. Try to disable tag verification on your S5. If it is disabled on S4 it works most of the time.

CheeseJuggernaut, 2018-02-07:
@Bombastisch No change when the verification is disabled on the S5. Is this a hardware issue or could I flash stock android onto my S5 to make it work

Bombastisch, 2018-02-07:
It has to do with the chipset so it is a hardware (bug). I don't think falshing a new rom will change it. (<- and this is not the solution we want to take. It's some bug in TagMo, but the devs are busy with rl, rn and it will take them a time to fix it. But you included a logcat so we will me able to fix it when there is time for it)

CheeseJuggernaut, 2018-02-07:
Alright cool. Let me know if you need testing done on any of these devices

Bombastisch, 2018-02-07:
You'll get a notification as soon they are back. I am sorry for the inconvenience. The old problem with Samsung phones really is a pain in the but, bc most of the time when somebody owns an Android it's one from Samsung...

CheeseJuggernaut, 2018-02-07:
I just got some new tags, and they worked first time. These tags are little disks: https://www.amazon.co.uk/dp/B075SBZ2NK/ref=cm_sw_r_cp_apa_Z-1EAb1WQH4AB

Even though buying new tags fixed my issue, I think it's important that we keep the ticket open until the original tags work or we figure out why they don't work. I've attached a scan of the new tags to see if any difference is visible from there.

[taginfo_scan_working.txt](https://github.com/HiddenRamblings/TagMo/files/1704629/taginfo_scan_working.txt)

HiddenRambler, 2018-02-07:
@CheeseJuggernaut 

The taginfo_scan[1].txt you attached. Is that a scan of a blank tag? The pages 4 to 10 are marked as locked (* +). If so that tag has already been locked somehow. 

TagMo only locks the tag as very last step and according to the log dump you attached it is failing when writing the main data (the first step). It is likely that the tags you brought first have are corrupted or already used.

If you compare it with the working scan you will see that the pages 4 to 129 are marked as unblocked (.) which should be the default unused NTAG.

CheeseJuggernaut, 2018-02-08:
Thanks, I've tested all of my old tags and they all have the exact same issue. I've brought up the issue with the vendor but they don't seem to understand the issue. I'm hoping to get a replacement set so that I can see if this is something that might be intentional

HiddenRambler, 2018-02-09:
Please see my last comment on #121

Bombastisch, 2018-02-09:
Maybe we should provide some links for working tags. What do you think @HiddenRambler ?  Can this be closed now?

CheeseJuggernaut, 2018-02-09:
Yea, this is solved. Faulty tags with pre-locked data.

closed 2018-02-09 by Bombastisch