HiddenRamblings/TagMo #121: Error while writing main data (stage 1) java.io.IOException: Transceive failed [Pixel 2 XL]

LazerScythe, 2018-02-08:
Getting this issue when trying to write on  [these cards] (https://www.amazon.co.uk/dp/B06Y489CZV/ref=pe_1909131_77697001_tnp_email_TE_AMZLdp_1) using a Pixel 2 XL and the same issue on my old Nexus 5X.

Reviews show the cards work for amiibos and NFCInfo shows this about the cards:

** TagInfo scan (version 4.23) 2018-02-08 15:48:40 **
Report Type: External

-- IC INFO ------------------------------

# IC manufacturer:
NXP Semiconductors

# IC type:
NTAG215

-- NDEF ------------------------------

# No NDEF data storage populated:

-- EXTRA ------------------------------

# Memory size:
504 bytes user memory
* 126 pages, with 4 bytes per page

# IC detailed information:
Full product name: NT2H1511G0DUx
Capacitance: 50 pF

# Version information:
Vendor ID: NXP
Type: NTAG
Subtype: 50 pF
Major version: 1
Minor version: V0
Storage size: 504 bytes
Protocol: ISO/IEC 14443-3

# Configuration information:
ASCII mirror disabled
NFC counter: disabled
No limit on wrong password attempts
Strong load modulation enabled

# Originality check:
Signature verified with NXP public key

-- FULL SCAN ------------------------------

# Technologies supported:
ISO/IEC 14443-3 (Type A) compatible
ISO/IEC 14443-2 (Type A) compatible

# Android technology information:
Tag description:
* TAG: Tech [android.nfc.tech.NfcA, android.nfc.tech.MifareUltralight, android.nfc.tech.NdefFormatable]
* Maximum transceive length: 253 bytes
* Default maximum transceive time-out: 618 ms

# Detailed protocol information:
ID: 04:50:6F:2A:4C:58:81
ATQA: 0x4400
SAK: 0x00

# Memory content:
[00] *  04:50:6F B3 (UID0-UID2, BCC0)
[01] *  2A:4C:58:81 (UID3-UID6)
[02] +  BF 48 53 45 (BCC1, INT, LOCK0-LOCK1)
[03] +  F1:10:FF:EE (OTP0-OTP3)
[04] *  03 00 FE 00 |....|
[05] +  00 00 00 00 |....|
[06] *  00 00 00 00 |....|
[07] +  00 00 00 00 |....|
[08] *  00 00 00 00 |....|
[09] +  00 00 00 00 |....|
[0A] x  00 00 00 00 |....|
[0B] .  00 00 00 00 |....|
[0C] .  00 00 00 00 |....|
[0D] .  00 00 00 00 |....|
[0E] x  00 00 00 00 |....|
[0F] .  00 00 00 00 |....|
[10] .  00 00 00 00 |....|
[11] .  00 00 00 00 |....|
[12] .  00 00 00 00 |....|
[13] .  00 00 00 00 |....|
[14] .  00 00 00 00 |....|
[15] .  00 00 00 00 |....|
[16] .  00 00 00 00 |....|
[17] .  00 00 00 00 |....|
[18] .  00 00 00 00 |....|
[19] .  00 00 00 00 |....|
[1A] .  00 00 00 00 |....|
[1B] .  00 00 00 00 |....|
[1C] .  00 00 00 00 |....|
[1D] .  00 00 00 00 |....|
[1E] .  00 00 00 00 |....|
[1F] .  00 00 00 00 |....|
[20] .  00 00 00 00 |....|
[21] .  00 00 00 00 |....|
[22] .  00 00 00 00 |....|
[23] .  00 00 00 00 |....|
[24] .  00 00 00 00 |....|
[25] .  00 00 00 00 |....|
[26] .  00 00 00 00 |....|
[27] .  00 00 00 00 |....|
[28] .  00 00 00 00 |....|
[29] .  00 00 00 00 |....|
[2A] .  00 00 00 00 |....|
[2B] .  00 00 00 00 |....|
[2C] .  00 00 00 00 |....|
[2D] .  00 00 00 00 |....|
[2E] .  00 00 00 00 |....|
[2F] .  00 00 00 00 |....|
[30] .  00 00 00 00 |....|
[31] .  00 00 00 00 |....|
[32] .  00 00 00 00 |....|
[33] .  00 00 00 00 |....|
[34] .  00 00 00 00 |....|
[35] .  00 00 00 00 |....|
[36] .  00 00 00 00 |....|
[37] .  00 00 00 00 |....|
[38] .  00 00 00 00 |....|
[39] .  00 00 00 00 |....|
[3A] .  00 00 00 00 |....|
[3B] .  00 00 00 00 |....|
[3C] .  00 00 00 00 |....|
[3D] .  00 00 00 00 |....|
[3E] .  00 00 00 00 |....|
[3F] .  00 00 00 00 |....|
[40] .  00 00 00 00 |....|
[41] .  00 00 00 00 |....|
[42] .  00 00 00 00 |....|
[43] .  00 00 00 00 |....|
[44] .  00 00 00 00 |....|
[45] .  00 00 00 00 |....|
[46] .  00 00 00 00 |....|
[47] .  00 00 00 00 |....|
[48] .  00 00 00 00 |....|
[49] .  00 00 00 00 |....|
[4A] .  00 00 00 00 |....|
[4B] .  00 00 00 00 |....|
[4C] .  00 00 00 00 |....|
[4D] .  00 00 00 00 |....|
[4E] .  00 00 00 00 |....|
[4F] .  00 00 00 00 |....|
[50] .  00 00 00 00 |....|
[51] .  00 00 00 00 |....|
[52] .  00 00 00 00 |....|
[53] .  00 00 00 00 |....|
[54] .  00 00 00 00 |....|
[55] .  00 00 00 00 |....|
[56] .  00 00 00 00 |....|
[57] .  00 00 00 00 |....|
[58] .  00 00 00 00 |....|
[59] .  00 00 00 00 |....|
[5A] .  00 00 00 00 |....|
[5B] .  00 00 00 00 |....|
[5C] .  00 00 00 00 |....|
[5D] .  00 00 00 00 |....|
[5E] .  00 00 00 00 |....|
[5F] .  00 00 00 00 |....|
[60] .  00 00 00 00 |....|
[61] .  00 00 00 00 |....|
[62] .  00 00 00 00 |....|
[63] .  00 00 00 00 |....|
[64] .  00 00 00 00 |....|
[65] .  00 00 00 00 |....|
[66] .  00 00 00 00 |....|
[67] .  00 00 00 00 |....|
[68] .  00 00 00 00 |....|
[69] .  00 00 00 00 |....|
[6A] .  00 00 00 00 |....|
[6B] .  00 00 00 00 |....|
[6C] .  00 00 00 00 |....|
[6D] .  00 00 00 00 |....|
[6E] .  00 00 00 00 |....|
[6F] .  00 00 00 00 |....|
[70] .  00 00 00 00 |....|
[71] .  00 00 00 00 |....|
[72] .  00 00 00 00 |....|
[73] .  00 00 00 00 |....|
[74] .  00 00 00 00 |....|
[75] .  00 00 00 00 |....|
[76] .  00 00 00 00 |....|
[77] .  00 00 00 00 |....|
[78] .  00 00 00 00 |....|
[79] .  00 00 00 00 |....|
[7A] .  00 00 00 00 |....|
[7B] .  00 00 00 00 |....|
[7C] .  00 00 00 00 |....|
[7D] .  00 00 00 00 |....|
[7E] .  00 00 00 00 |....|
[7F] .  00 00 00 00 |....|
[80] .  00 00 00 00 |....|
[81] .  00 00 00 00 |....|
[82] .  00 00 00 BD (LOCK2-LOCK4, CHK)
[83] .  04 00 00 FF (CFG, MIRROR, AUTH0)
[84] .  00 05 -- -- (ACCESS)
[85] +P FF FF FF FF (PWD0-PWD3)
[86] +P 00 00 -- -- (PACK0-PACK1)

  *:locked & blocked, x:locked,
  +:blocked, .:un(b)locked, ?:unknown
  r:readable (write-protected),
  p:password protected, -:write-only
  P:password protected write-only

--------------------------------------

HiddenRambler, 2018-02-09:
The scan you included there is it of a brand new card not used with TagMo? if so then it's been corrupted/written to already.

The last two numbers in the third line in the memory content which in your card shows:
[02] + BF 48 53 45 (BCC1, INT, LOCK0-LOCK1)

Should on a brand new card should show as below:

[02] + BF 48 00 00 (BCC1, INT, LOCK0-LOCK1)

essentially your card have already locked part of the tag from writing. Off the top of my head I do not remember if these bytes can be changed once written. I will need to look it up. If it can be changed then I will try to include a feature to recover them.

If the above dump was of a brand new blank tag then I'd suggest you contact the seller. It looks like there are others with the same issue #117  #118

HiddenRambler, 2018-02-09:
Bad news, I just tried deliberately setting the LOCK0/1 bytes to match your card and then trying to reset it to the default. It looks like it's not allowed as they seem to be one time programmable bytes.

TagMo would not cause this type of data to be written even when something goes wrong in writing so I do not believe this was caused by TagMo. like i've mentioned in the other two related issues I think there is a bad batch of NTAGs going out. Best option is to return it to the seller as defective.

LazerScythe, 2018-02-09:
Thanks a lot for the info. I sent back my cards and ordered stickers this time, reviews look good for ammiibo use.

The linked cards have some reviews of working with amiibos, wonder if the manufacturer changed something.

closed 2018-02-09 by LazerScythe