HiddenRamblings/TagMo #122: Inconsistent Success rates. Only 7 out of 20 tags (NFC215) wrote to without error.

Raynr, 2018-02-16:
I'm using an Android phone, an ZTE Axon 7 (a chinese phone manufacturer) (Model ZTE 2017U, Build #: A2017UV1.1.0B35), android version 7.1.1, using TagMo build 2.6.1.

While my first few attempts worked closer to 1 out of 4 tries. Following your steps exactly, although usually but not the last 3 times I tried, I place the blank tag underneath the NFC sensor before I click "Write", I'm very careful by having my phone lay on my desk, so it doesn't move when I tap the "write" button, I still only succeed occassionaly. Usually I recieve a "Tag Lost" error, or that "Error while writing main data (stage 1) java.io.IOException: Transceive failed", but that one is rarer.

I've checked out all the listed issues that seemed similar to mine, but the all seemed to be 100% failure until you fixed them, unlike my 75-85% failure rate.

Do you know what could be causing this, and what more info do you need?
*I'm also a developer, but usually of websites and desktop apps, and I'm willing to help out on a more complex level if needed.

Bombastisch, 2018-02-16:
Hi @Rayner. Are you sure your tags are working? "Tag lost" only means you didn't hold the Tag over the NFC reader/writer correctly.  Error while writing often (but not neccessarily) means that some bytes of the tgas memory are locked (maybe by the seller) and they won't work at all and you might need to ask for a refund, since they are "broken". As a first step, since you said "lost tag" occured more often,  maybe you have a phone case put on your phone which blocks the writing process. If you have one remove it and try again.  

Otherwise please download "NXP taginfo" from the playstore and post a fullscan of one of your tag in this issue, which didn't work. Maybe we can detect if some bytes of your tag are blocked.

Raynr, 2018-02-16:
Alright, I'll download the NXP tagingo app.
I did have the case on for the first 10 I tried, but I relaized it might be an issue and took it off for the next set of ten.
All these tags were from the same dealer. : https://www.amazon.ca/gp/product/B078T752SL/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1

Before each attempt, I let my phone just get triggered by it on it's own, so I'd see the "new blank tag" pop-up first.  Sometimes, I'd have to wait a bit because while letting the fresh blank tag sit under my phone, the alert would pop-up multiple times. So I'd move it around until it would only pop-up once, and then try writing to it.

Bombastisch, 2018-02-16:
Could you use the "taginfo" app. Open up the menu on the left (top left button) and switch to "full scan" . Put the tag under your phone and let it scan. Afterwards click on the share icon -> copy to clipboard and send the text in here. Ty :)

purpldevl, 2018-02-16:
I'm using an Axon 7 and the first few times I attempted to write I ran into the same issue - make sure you're using the proper space for the NFC reader / writer on the phone. It's located right above the camera, right about where the case has a "T" shape.

Bombastisch, 2018-02-16:
Has this worked for you @Raynr

CTCaer, 2018-02-17:
@Raynr 
Check [this ](https://youtu.be/s1uFZHSWU4c?t=202)

If you pause the video, you will see that as @purpldevl mentioned, the NFC coil is around the camera.
Because the phone back cover is made out of aluminum, you have to be exact with the tag placement.
Most of the energy needed to power the tag or to get data from the tag, is absorbed by the metal case.
So you need to place the tag at center of the camera.

(In the video, don't be confused with the big black rectangle. That's actually a masking tape that keeps the fingerprint in place, provides somewhat water resistance and some padding for the phone to feel solid.)

Raynr, 2018-02-17:
Here's the Full Nxp tagine scan from the latest. 

** TagInfo scan (version 4.23) 2018-02-16 22:23:39 **
Report Type: External

-- IC INFO ------------------------------

# IC manufacturer:
NXP Semiconductors

# IC type:
NTAG215

-- NDEF ------------------------------

# No NDEF Message present:

-- EXTRA ------------------------------

# Memory size:
504 bytes user memory
* 126 pages, with 4 bytes per page

# IC detailed information:
Full product name: NT2H1511G0DUx
Capacitance: 50 pF

# Version information:
Vendor ID: NXP
Type: NTAG
Subtype: 50 pF
Major version: 1
Minor version: V0
Storage size: 504 bytes
Protocol: ISO/IEC 14443-3

# Configuration information:
ASCII mirror disabled
NFC counter: disabled
No limit on wrong password attempts
Strong load modulation enabled

# Originality check:
Signature verified with NXP public key

-- FULL SCAN ------------------------------

--------------------------------------

Bombastisch, 2018-02-17:
@Raynr the "Full Scan" is missing. Swipe to the very right of the app and do a full scan so it dumps every byte of the tag. This is the part we need to investigate wether it's your phone or the tags are failing

Raynr, 2018-02-17:
This is the best I could get, kept giving me "scan interrupted" errors. 

** TagInfo scan (version 4.23) 2018-02-16 22:54:21 **
Report Type: External

-- IC INFO ------------------------------

# IC manufacturer:
NXP Semiconductors

# IC type:
NTAG215

-- NDEF ------------------------------

# No NDEF Message present:

-- EXTRA ------------------------------

# Memory size:
504 bytes user memory
* 126 pages, with 4 bytes per page

# IC detailed information:
Full product name: NT2H1511G0DUx
Capacitance: 50 pF

# Version information:
Vendor ID: NXP
Type: NTAG
Subtype: 50 pF
Major version: 1
Minor version: V0
Storage size: 504 bytes
Protocol: ISO/IEC 14443-3

# Configuration information:
ASCII mirror disabled
NFC counter: disabled
No limit on wrong password attempts
Strong load modulation enabled

# Originality check:
Signature verified with NXP public key

-- FULL SCAN ------------------------------

# Technologies supported:
ISO/IEC 14443-3 (Type A) compatible
ISO/IEC 14443-2 (Type A) compatible

# Android technology information:
Tag description:
* TAG: Tech [android.nfc.tech.NfcA, android.nfc.tech.MifareUltralight, android.nfc.tech.Ndef]
* Maximum transceive length: 253 bytes
* Default maximum transceive time-out: 618 ms

# Detailed protocol information:
ID: 04:F6:9E:3A:61:57:80
ATQA: 0x4400
SAK: 0x00

# Memory content:
[00] *  04:F6:9E E4 (UID0-UID2, BCC0)
[01] *  3A:61:57:80 (UID3-UID6)
[02] .  8C 48 00 00 (BCC1, INT, LOCK0-LOCK1)
[03] .  E1:10:3E:00 (OTP0-OTP3)
[04] .  03 00 FE 00 |....|
[05] .  00 00 00 00 |....|
[06] .  00 00 00 00 |....|
[07] .  00 00 00 00 |....|
[08] .  00 00 00 00 |....|
[09] .  00 00 00 00 |....|
[0A] .  00 00 00 00 |....|
[0B] .  00 00 00 00 |....|
[0C] .  00 00 00 00 |....|
[0D] .  00 00 00 00 |....|
[0E] .  00 00 00 00 |....|
[0F] .  00 00 00 00 |....|
[10] ?  00 00 00 00 |....|
[11] ?  00 00 00 00 |....|
[12] ?  00 00 00 00 |....|
[13] ?  00 00 00 00 |....|

<aborted: tag disappeared>

--------------------------------------

Bombastisch, 2018-02-17:
Scan interupted errors in this app and in TagMo, seems like something is blocking your NFC Reader/Writer or it's broken. Do you have a friends/ secondary phone to retry writing to your tags from another phone?

Raynr, 2018-02-20:
Not yet, I'll have to convince one of them to try Tagmo out too.
Most of my friends either live an inconvienient distance away for this kind of thing, use iPhones, or work the opposite kind of hours as I do.

Bombastisch, 2018-02-20:
Did you have any more success with this? You tags seem fine (but we can't say fo sure until somebody else tried to use them)

Raynr, 2018-02-20:
No, I only have 2 blank tags left, so I haven't risked ruining anymore until I find a new NFC device, or my next order of tags comes in (which I have yet to order).

HiddenRambler, 2018-02-20:
From everything said in the this thread it sounds like a issue with your phone losing connection with the tag while writing. TagMo tries to write in a very conservative manner which means that the tags that have gone bad might be recoverable. But first you'd have to fix make sure that the phone NFC works in a stable fashion.

I think the key here is to figure out the sweetspot where you need to hold the tag on the phone. It might be easier to laydown the phone on it's display and then to place the tag on it. Keep doing this until you can realiably read a tag with the NXP TagInfo app without any errors.

I would not recommend you attempt to write any tag until you can do this.

purpldevl, 2018-02-20:
Adding on to this, take off any case you might have on it. I've gotten great results by holding the phone on the tag right where the phone starts to bend at the top, directly in the center. I had a rubber case on it for a bit that seemed to mess with writing, but I've had successful writes since removing it.

zombiederf, 2018-02-25:
I have a moto z2 force. nice aluminum back. big camera screen. blah blah blah....100% tag failing to read, do anything, etc, etc.....for shits cuz it is a 2cm field. tried it on front of my screen....worked just fine. FFS.
but it worked. try it if you haven't had luck.

Bombastisch, 2018-03-14:
@Raynr did you have any success or did you ask a friend? I hope we find a solution to this :)

Raynr, 2018-03-14:
@Bombastisch , I have yet to find a friend willing to install this on their phone for me. :(
Currently, I've settled with using the 7 amiibo tags I made already.

Bombastisch, 2020-01-17:
Maybe try using the newest version and see if it works now

AbandonedCart, 2021-09-11:
This appears to have been solved, but please comment if it is still an issue and needs to be opened again.

closed 2021-09-11 by AbandonedCart