HiddenRamblings/TagMo #123: browsing for keys only shows picture and music app options?

Nanashi87, 2018-03-05:
Sgh-t999l tagmo app opens and says keys not loaded and when you click them to load them it doesn't have normal file browsing, it only shows google music, dropbox, gallery, music player, and photos.

Bombastisch, 2018-03-05:
Can you send a screenshot from that file browser which shows up, when you try to load the keys? Usually you can open a overview with the button on the top left.

Nanashi87, 2018-03-07:
I figured out  how to work around it with dropbox but here's what it shows

DimensionsGaming, 2018-03-13:
I had that problem before, but what I did to fix it was I installed a file explorer app from the play store, because it doesn't look as if the app selector sees the built in file explorer.

Bombastisch, 2018-03-18:
I opene an enhancement request, so maybe we get a file browser to circumvent that problem. The request is here: #126

Are you guys fine with closing this for now?

Bombastisch, 2018-06-04:
I think the issue is solved.  Request #126 is still open. My programming skills are sadly still not existent. :/

Maybe Hiddenrambler or North101 will return one day with a lot of new energy go push this project forward.
We still have a new UI that's not done yet, but it will improve TagMo a lot once finished. :)

closed 2018-06-04 by Bombastisch