HiddenRamblings/TagMo #125: Consider submitting app to Fdroid

Necklaces, 2018-03-13:
Not sure what license this repository is under - would be nice with a clarification, but if it's any of the F/LOSS licenses you have the possibility of adding this app to the [Fdroid](https://f-droid.org/) so people can more easily install the app. What do you think?

jayayou, 2018-11-18:
I've been browsing on this project for 2 hours, and I can't figure out how to install on my phone

Necklaces, 2018-11-18:
Well then, since we've already been necrobumped by an off topic question, I'll take the time to remind everyone about the consequences of not having a license:

https://choosealicense.com/no-permission/

If the owner(s) of this project want to submit a request for packaging to Fdroid, see:

https://gitlab.com/fdroid/rfp

Bombastisch, 2018-11-18:
Thank you for you interest in TagMo.

I wrote the owner of this repo about adding a license on Febuary 22 (according to discord).

Sadly I haven't recieved any answer yet. I am really sad that this awesome project fades out and I really wish there would be a comeback.

Until then we will have to wait and hope we finally get a license on this. :)

fargeble, 2019-08-19:
According to https://github.com/HiddenRamblings/TagMo/blob/master/app/src/main/java/com/hiddenramblings/tagmo/TagUtil.java#L15 there is GPL v2 code copied into the project, so presumably it should all be GPL. The origin can be found at https://github.com/codlab/amiibo/blob/master/app/src/main/java/eu/codlab/amiiwrite/amiibo/AmiiboMethods.java#L21

AbandonedCart, 2021-10-16:
Opting for a temporary GNU license, until such time as there are any objections. Given the terms of the GNU license, it would allow a more restrictive license to replace it later.