HiddenRamblings/TagMo #127: Error while writing main data (stage 1) java.io.IOExeption: transience failed

Lionel11, 2018-03-17:
Hello,
i have this error when i try to write on my tags. I downloaded  NXP tag info, it's NTAG215, i have a huawei P8 lite on android 7.0, i tried on many version but its never work, can you help me please ? (i already deactivate tag validation)

Lionel11, 2018-03-17:
i have been tested with a samsung S4, S5, huawei p8 lite, p10 lite... with some versions... impossible to write always the same error

Lionel11, 2018-03-17:
i buyed nfc cards where guys mentionned it was good for write amiibo,

Lionel11, 2018-03-17:
so i think cards are ok

Bombastisch, 2018-03-17:
@Lionel11 

As far as I can tell it's something about the cards. Can you post screenshots of the NXP taginfo app when you scan your tags? Also include the "Full scan" (it's when you swipe to the very right).

Lionel11, 2018-03-17:
ok i will do that

Lionel11, 2018-03-17:
[image]
[image]
[image]

Bombastisch, 2018-03-17:
Ur screenshot doesn't include the "real" full scan. Maybe you uncheckdled fullscan in the options. If you click on the 3 yellow stripes in the top left a menu will show up and please select full scan. Then read the tag and in the section "full scan" there will be a table with lot's of numbers. Please screenshot that, because it gives us more detailed data about your tag

Lionel11, 2018-03-17:
[image]

Bombastisch, 2018-03-17:
Yes that's it and I got bad news for you. :( Your tags have been prelocked. If you look on the left side in the first rows of the bytes are locked and/or blocked.

Apperantly your seller got you prelocked tags.

I don't know where you've bought them, but usually you can return them and get your money back.

Be sure to read comments under the product if users have had success using them (Watch out: Amazon comments are getting often mixed up)

Lionel11, 2018-03-17:
Thank you for your response, what is a "prelocked tag" ? is it normal ?  i bought them on amazon

Bombastisch, 2018-03-17:
[image]

Bytes in row 04 - 0A are locked. 

Description is here:

[image]

These bytes shouldn't be locked. Some sellers have a few bytes locked and then TagMo can't write to it. Thus the error message popping up.

This isn't normal. Maybe write the seller about his tags. They can't be written -> No Amiibo

Return your tags on Amazon and you get your money back.

Lionel11, 2018-03-17:
Thank you very much. So i need to have the parameter un(b)ocked ? but on which line exactly ?

Bombastisch, 2018-03-27:
A working tag should look similar to this. The middle part needs to be completly "." marked. So you can write to the tag.

** TagInfo scan 

# Memory content:
[00] *  04:89:BB BE (UID0-UID2, BCC0)
[01] *  A2:5B:57:80 (UID3-UID6)
[02] .  2E 48 00 00 (BCC1, INT, LOCK0-LOCK1)
[03] .  E1:10:3E:00 (OTP0-OTP3)
[04] .  03 00 FE 00 |....|
[05] .  00 00 00 00 |....|
[06] .  00 00 00 00 |....|
[07] .  00 00 00 00 |....|
[08] .  00 00 00 00 |....|
[09] .  00 00 00 00 |....|
[0A] .  00 00 00 00 |....|
[0B] .  00 00 00 00 |....|
[0C] .  00 00 00 00 |....|
[0D] .  00 00 00 00 |....|
[0E] .  00 00 00 00 |....|
[0F] .  00 00 00 00 |....|
[10] .  00 00 00 00 |....|
[11] .  00 00 00 00 |....|
[12] .  00 00 00 00 |....|
[13] .  00 00 00 00 |....|
[14] .  00 00 00 00 |....|
[15] .  00 00 00 00 |....|
[16] .  00 00 00 00 |....|
[17] .  00 00 00 00 |....|
[18] .  00 00 00 00 |....|
[19] .  00 00 00 00 |....|
[1A] .  00 00 00 00 |....|
[1B] .  00 00 00 00 |....|
[1C] .  00 00 00 00 |....|
[1D] .  00 00 00 00 |....|
[1E] .  00 00 00 00 |....|
[1F] .  00 00 00 00 |....|
[20] .  00 00 00 00 |....|
[21] .  00 00 00 00 |....|
[22] .  00 00 00 00 |....|
[23] .  00 00 00 00 |....|
[24] .  00 00 00 00 |....|
[25] .  00 00 00 00 |....|
[26] .  00 00 00 00 |....|
[27] .  00 00 00 00 |....|
[28] .  00 00 00 00 |....|
[29] .  00 00 00 00 |....|
[2A] .  00 00 00 00 |....|
[2B] .  00 00 00 00 |....|
[2C] .  00 00 00 00 |....|
[2D] .  00 00 00 00 |....|
[2E] .  00 00 00 00 |....|
[2F] .  00 00 00 00 |....|
[30] .  00 00 00 00 |....|
[31] .  00 00 00 00 |....|
[32] .  00 00 00 00 |....|
[33] .  00 00 00 00 |....|
[34] .  00 00 00 00 |....|
[35] .  00 00 00 00 |....|
[36] .  00 00 00 00 |....|
[37] .  00 00 00 00 |....|
[38] .  00 00 00 00 |....|
[39] .  00 00 00 00 |....|
[3A] .  00 00 00 00 |....|
[3B] .  00 00 00 00 |....|
[3C] .  00 00 00 00 |....|
[3D] .  00 00 00 00 |....|
[3E] .  00 00 00 00 |....|
[3F] .  00 00 00 00 |....|
[40] .  00 00 00 00 |....|
[41] .  00 00 00 00 |....|
[42] .  00 00 00 00 |....|
[43] .  00 00 00 00 |....|
[44] .  00 00 00 00 |....|
[45] .  00 00 00 00 |....|
[46] .  00 00 00 00 |....|
[47] .  00 00 00 00 |....|
[48] .  00 00 00 00 |....|
[49] .  00 00 00 00 |....|
[4A] .  00 00 00 00 |....|
[4B] .  00 00 00 00 |....|
[4C] .  00 00 00 00 |....|
[4D] .  00 00 00 00 |....|
[4E] .  00 00 00 00 |....|
[4F] .  00 00 00 00 |....|
[50] .  00 00 00 00 |....|
[51] .  00 00 00 00 |....|
[52] .  00 00 00 00 |....|
[53] .  00 00 00 00 |....|
[54] .  00 00 00 00 |....|
[55] .  00 00 00 00 |....|
[56] .  00 00 00 00 |....|
[57] .  00 00 00 00 |....|
[58] .  00 00 00 00 |....|
[59] .  00 00 00 00 |....|
[5A] .  00 00 00 00 |....|
[5B] .  00 00 00 00 |....|
[5C] .  00 00 00 00 |....|
[5D] .  00 00 00 00 |....|
[5E] .  00 00 00 00 |....|
[5F] .  00 00 00 00 |....|
[60] .  00 00 00 00 |....|
[61] .  00 00 00 00 |....|
[62] .  00 00 00 00 |....|
[63] .  00 00 00 00 |....|
[64] .  00 00 00 00 |....|
[65] .  00 00 00 00 |....|
[66] .  00 00 00 00 |....|
[67] .  00 00 00 00 |....|
[68] .  00 00 00 00 |....|
[69] .  00 00 00 00 |....|
[6A] .  00 00 00 00 |....|
[6B] .  00 00 00 00 |....|
[6C] .  00 00 00 00 |....|
[6D] .  00 00 00 00 |....|
[6E] .  00 00 00 00 |....|
[6F] .  00 00 00 00 |....|
[70] .  00 00 00 00 |....|
[71] .  00 00 00 00 |....|
[72] .  00 00 00 00 |....|
[73] .  00 00 00 00 |....|
[74] .  00 00 00 00 |....|
[75] .  00 00 00 00 |....|
[76] .  00 00 00 00 |....|
[77] .  00 00 00 00 |....|
[78] .  00 00 00 00 |....|
[79] .  00 00 00 00 |....|
[7A] .  00 00 00 00 |....|
[7B] .  00 00 00 00 |....|
[7C] .  00 00 00 00 |....|
[7D] .  00 00 00 00 |....|
[7E] .  00 00 00 00 |....|
[7F] .  00 00 00 00 |....|
[80] .  00 00 00 00 |....|
[81] .  00 00 00 00 |....|
[82] .  00 00 00 BD (LOCK2-LOCK4, CHK)
[83] .  04 00 00 FF (CFG, MIRROR, AUTH0)
[84] .  00 05 -- -- (ACCESS)
[85] +P FF FF FF FF (PWD0-PWD3)
[86] +P 00 00 -- -- (PACK0-PACK1)

  *:locked & blocked, x:locked,
  +:blocked, .:un(b)locked, ?:unknown
  r:readable (write-protected),
  p:password protected, -:write-only
  P:password protected write-only

--------------------------------------

Bombastisch, 2018-03-27:
In your case line 04 - 0A are prelocked and so TagMo can't write to it.

Bombastisch, 2018-06-04:
Did you have luck with some new tags?

Lionel11, 2018-06-04:
Hello, yes i have ordered new tags and it works ^^

Bombastisch, 2018-06-04:
That's awesome! I think I can savely close this now. Have fun with TagMo :)

closed 2018-06-04 by Bombastisch