HiddenRamblings/TagMo #129: Also getting Error while writing main data (stage 1) java.io.IOExeption: transience failed

Spontant, 2018-03-24:
I'm also getting Error while writing main data (stage 1) java.io.IOExeption: transience failed but from what I can gather I don't think it's the stickers I have, unless I don't understand what I am looking at in the NFC scans.
These are the results of the scan, could you please tell me if my tags are ok and where the problem might be?

** TagInfo scan (version 4.23) 2012-05-24 22:35:54 **
Report Type: External

**-- IC INFO ------------------------------**

# IC manufacturer:
NXP Semiconductors

# IC type:
NTAG215

**-- NDEF ------------------------------**

# No NDEF Message present:

**-- EXTRA ------------------------------**

# Memory size:
504 bytes user memory
* 126 pages, with 4 bytes per page

# IC detailed information:
Full product name: NT2H1511G0DUx
Capacitance: 50 pF

# Version information:
Vendor ID: NXP
Type: NTAG
Subtype: 50 pF
Major version: 1
Minor version: V0
Storage size: 504 bytes
Protocol: ISO/IEC 14443-3

# Configuration information:
ASCII mirror disabled
NFC counter: disabled (no tearing)
No limit on wrong password attempts
Strong load modulation enabled

# Originality check:
Signature verified with NXP public key

**-- FULL SCAN ------------------------------**

# Technologies supported:
ISO/IEC 14443-3 (Type A) compatible
ISO/IEC 14443-2 (Type A) compatible

# Android technology information:
Tag description:
* TAG: Tech [android.nfc.tech.NfcA, android.nfc.tech.MifareUltralight, android.nfc.tech.Ndef, android.nfc.tech.NdefFormatable]
* Maximum transceive length: 253 bytes
* Default maximum transceive time-out: [not available]

# Detailed protocol information:
ID: 04:AE:22:02:F9:59:80
ATQA: 0x4400
SAK: 0x00

# Memory content:
[00] *  04:AE:22 00 (UID0-UID2, BCC0)
[01] *  02:F9:59:80 (UID3-UID6)
[02] .  22 48 00 00 (BCC1, INT, LOCK0-LOCK1)
[03] .  E1:10:3E:00 (OTP0-OTP3)
[04] .  03 00 FE 00 |....|
[05] .  00 00 00 00 |....|
[06] .  00 00 00 00 |....|
[07] .  00 00 00 00 |....|
[08] .  00 00 00 00 |....|
[09] .  00 00 00 00 |....|
[0A] .  00 00 00 00 |....|
[0B] .  00 00 00 00 |....|
[0C] .  00 00 00 00 |....|
[0D] .  00 00 00 00 |....|
[0E] .  00 00 00 00 |....|
[0F] .  00 00 00 00 |....|
[10] ?  00 00 00 00 |....|
[11] ?  00 00 00 00 |....|
[12] ?  00 00 00 00 |....|
[13] ?  00 00 00 00 |....|
[14] ?  00 00 00 00 |....|
[15] ?  00 00 00 00 |....|
[16] ?  00 00 00 00 |....|
[17] ?  00 00 00 00 |....|
[18] ?  00 00 00 00 |....|
[19] ?  00 00 00 00 |....|
[1A] ?  00 00 00 00 |....|
[1B] ?  00 00 00 00 |....|
[1C] ?  00 00 00 00 |....|
[1D] ?  00 00 00 00 |....|
[1E] ?  00 00 00 00 |....|
[1F] ?  00 00 00 00 |....|
[20] ?  00 00 00 00 |....|
[21] ?  00 00 00 00 |....|
[22] ?  00 00 00 00 |....|
[23] ?  00 00 00 00 |....|
[24] ?  00 00 00 00 |....|
[25] ?  00 00 00 00 |....|
[26] ?  00 00 00 00 |....|
[27] ?  00 00 00 00 |....|
[28] ?  00 00 00 00 |....|
[29] ?  00 00 00 00 |....|
[2A] ?  00 00 00 00 |....|
[2B] ?  00 00 00 00 |....|
[2C] ?p XX XX XX XX
[2D] ?p XX XX XX XX
[2E] ?p XX XX XX XX
[2F] ?p XX XX XX XX
[30] ?p XX XX XX XX
[31] ?p XX XX XX XX
[32] ?p XX XX XX XX
[33] ?p XX XX XX XX
[34] ?p XX XX XX XX
[35] ?p XX XX XX XX
[36] ?p XX XX XX XX
[37] ?p XX XX XX XX
[38] ?p XX XX XX XX
[39] ?p XX XX XX XX
[3A] ?p XX XX XX XX
[3B] ?p XX XX XX XX
[3C] ?p XX XX XX XX
[3D] ?p XX XX XX XX
[3E] ?p XX XX XX XX
[3F] ?p XX XX XX XX
[40] ?p XX XX XX XX
[41] ?p XX XX XX XX
[42] ?p XX XX XX XX
[43] ?p XX XX XX XX
[44] ?p XX XX XX XX
[45] ?p XX XX XX XX
[46] ?p XX XX XX XX
[47] ?p XX XX XX XX
[48] ?p XX XX XX XX
[49] ?p XX XX XX XX
[4A] ?p XX XX XX XX
[4B] ?p XX XX XX XX
[4C] ?p XX XX XX XX
[4D] ?p XX XX XX XX
[4E] ?p XX XX XX XX
[4F] ?p XX XX XX XX
[50] ?p XX XX XX XX
[51] ?p XX XX XX XX
[52] ?p XX XX XX XX
[53] ?p XX XX XX XX
[54] ?p XX XX XX XX
[55] ?p XX XX XX XX
[56] ?p XX XX XX XX
[57] ?p XX XX XX XX
[58] ?p XX XX XX XX
[59] ?p XX XX XX XX
[5A] ?p XX XX XX XX
[5B] ?p XX XX XX XX
[5C] ?p XX XX XX XX
[5D] ?p XX XX XX XX
[5E] ?p XX XX XX XX
[5F] ?p XX XX XX XX
[60] ?p XX XX XX XX
[61] ?p XX XX XX XX
[62] ?p XX XX XX XX
[63] ?p XX XX XX XX
[64] ?p XX XX XX XX
[65] ?p XX XX XX XX
[66] ?p XX XX XX XX
[67] ?p XX XX XX XX
[68] ?p XX XX XX XX
[69] ?p XX XX XX XX
[6A] ?p XX XX XX XX
[6B] ?p XX XX XX XX
[6C] ?p XX XX XX XX
[6D] ?p XX XX XX XX
[6E] ?p XX XX XX XX
[6F] ?p XX XX XX XX
[70] ?p XX XX XX XX
[71] ?p XX XX XX XX
[72] ?p XX XX XX XX
[73] ?p XX XX XX XX
[74] ?p XX XX XX XX
[75] ?p XX XX XX XX
[76] ?p XX XX XX XX
[77] ?p XX XX XX XX
[78] ?p XX XX XX XX
[79] ?p XX XX XX XX
[7A] ?p XX XX XX XX
[7B] ?p XX XX XX XX
[7C] ?p XX XX XX XX
[7D] ?p XX XX XX XX
[7E] ?p XX XX XX XX
[7F] ?p XX XX XX XX
[80] ?p XX XX XX XX
[81] ?p XX XX XX XX
[82] ?p XX XX XX XX (LOCK2-LOCK4, CHK)
[83] ?p XX XX XX XX (CFG, MIRROR, AUTH0)
[84] ?p XX XX -- -- (ACCESS)
[85] +P XX XX XX XX (PWD0-PWD3)
[86] +P XX XX -- -- (PACK0-PACK1)

  *:locked & blocked, x:locked,
  +:blocked, .:un(b)locked, ?:unknown
  r:readable (write-protected),
  p:password protected, -:write-only
  P:password protected write-only

--------------------------------------

Bombastisch, 2018-03-24:
Hi, looking at the tag binary data some of you bytes seem to be password locked. With the app "NFC tagwriter by NXP"on the Google Playstore you can try to erase your tag and reset it completly. Hopefully the password protection goes away and tagmo can write to it. Tell me if it worked :)

Spontant, 2018-03-25:
I tried it and when I try to write to the tag using tagmo it says tag detected and goes on forever without doing anything.

Bombastisch, 2018-03-25:
Could you provide a TagInfo Scan again?

Spontant, 2018-03-25:
After I try to erase it or after I try to write to it?

Bombastisch, 2018-03-25:
After the factory reset

Spontant, 2018-03-25:
Yup, the one I sent is after the factory reset.

Bombastisch, 2018-03-25:
The full scan isn't included :/  i need the table with the binary data. :)

Spontant, 2018-03-25:
After I factory reset it, that's all the scan gives me. It doesn't show anything else, I try to scroll down and it doesn't show anything else, the binary data doesn't show up at all. It's very strange, because I am in full scan mode and it doesn't show it at all.

Bombastisch, 2018-03-25:
Can you then try to capture a logcat? Try to write to it in TagMo and then skip the writing process and click "save logcat" in the top left "3 button" menu. I can't read this, tho. @HiddenRambler or @North101 can check that.

//--/  Do all your tags show this problem?

Spontant, 2018-03-25:
[tagmo_logcat.txt](https://github.com/HiddenRamblings/TagMo/files/1845727/tagmo_logcat.txt)

This is the logcat it gives me, and from what I can tell all of the tags do this, so they might just be defective.
Thank you for trying to help me by the way. :)

markthewolf90, 2019-02-07:
I know it's an old issue, but I'm having the same problem with all of my tags. It says either "tag was lost" or "transceive failed" on repeat when I try to write, and when I try to scan a blank tag or a tag that I've written onto before, it says "tag was lost" over and over again. The app and the tags were working for me last year, the only thing that's changed since then is that my phone updated from Android 7.0 to 8.0 a few months back. (Which did cause some apps to become corrupted, actually. But I only just installed TagMo today.)

Bombastisch, 2019-02-07:
Maybe try to remove your phone case and place the tag at the exact spot of the NFC reader? You phone's manual may help.

Bombastisch, 2019-02-08:
Have you tried disabling the Tag Validation in TagMo's settings?

Which phone do you use?

closed 2020-01-31 by Bombastisch