HiddenRamblings/TagMo #130: SSB Data is limited

LeafBlitz, 2018-03-26:
Any amiibo that has a character in SSB can be used as an SSB amiibo, but only SSB series amiibo can be edited.

Bombastisch, 2018-03-26:
I have seen that on Reddit yesterday as well. @North101 's new beta apk fixes that. But this is still underdevelopment and TagMo is a free project and the devs have to work irl as well, it takes a time until it will be ready to be merged and an official release.

LeafBlitz, 2018-03-26:
The link you posted says the file has been removed, but thanks anyways!

Bombastisch, 2018-03-26:
Welp. @North101 removed it then. You can still compile the .apk from the "experimental ui" branch in Android Studio and test it.

AbandonedCart, 2021-09-11:
This has become a part of the released experimental branch. Thanks @North101 for all your hard work.

closed 2021-09-11 by AbandonedCart