HiddenRamblings/TagMo #132: Stuck in Tag Detected

RyuuzakiJulio, 2018-06-11:
I have a Sony Xperia (from Japan, language settings are in English, has no phone service, just wifi)
Have installed all versions (even beta ones, one by one and deleting cache and uninstalling previous app before installing new ones)
The tags I got I purchased from Amazon as "NTAG215 Nintendo NFC 215 Cards ami-bo Amiibo tagmo NFC Card with Built-in NFC Chip and similar Blank PVC ISO Card with NFC Tags Android and All of Your Phone, NFC Enabled" from a seller in Japan, reviewers say they have made 10/10 Amiibos with this item.

As for the installed version of TAGMO some versions simply just don't load anything, but the latest versions, everything goes well until I click WRITE TAG, then it starts scanning and when I place one of this new cards it does say TAG DETECTED, but then it just keeps "scanning" and it doesn't actually do anything.

I tried another tag writing app just to write a website into one of this tags and it was successful in a matter of seconds. (Of course making that tag unusable anymore)

When the app is not running, and I scan an empty tag, the phone immediately recognizes that is an empty tag and suggests I use a writing app to put info in it. Tried with other apps and works fine. But not any version of TagMo.

Any is it the Tag? the Phone? or the App? I don't have access to any android phone to test. I would have to go buy a used one, but then I don't know which phones from Japan work.

RyuuzakiJulio, 2018-06-12:
Sadly no, I don't own any Amiibo, I just got the Switch and wanted to try out that scan feature, but looking at all the different Amiibos out there, is something my wallet won't forgive me to get.
The working versions of TagMo do detect that there is a tag but that's it.

Bombastisch, 2018-06-12:
Can you check one of your blank tags like I described in #127 ? Then we can find out if it's the tags or not. Thus rule out the tags if they are working correctly. :)

RyuuzakiJulio, 2018-06-12:
I went ahead and got a different set of tags from another seller, also saying they work for Amiibo, I found that the author made a 3DS app. I was successful into getting my New 3DS to write all the tags no problem and they do work for the Switch. So apparently the tags weren’t an issue.

RyuuzakiJulio, 2018-06-12:
I downloaded another app on the phone to write just normal tags like website or contact info, and it worked no problem. TagMo only detected that there was a tag but it wasn’t reading or writing anything.

Bombastisch, 2018-06-13:
Just to clarify, you have loaded the keys in TagMo?

Do you know your .bin files are valid ones?

(Please don't post them here, just be sure the .bin files work, usually if TagMo can load them and you can view them in Hex, they work)

Bombastisch, 2019-06-27:
I assume it is probably the same bug as in #149  ? There was a solution proposed. Can't check myself if it works tho

Bombastisch, 2020-01-17:
Should be fixed in newest version!

closed 2020-01-17 by Bombastisch