HiddenRamblings/TagMo #133: “Failed to decrypt tag” error

askep3, 2018-08-08:
I have a galaxy s5 on the latest version of Tagmo, and am using a popular bin file that is verified to work, but keep getting this error when attempting to write the tag. I am sure I am using NTAG215 tags since I used the TagInfo app to check. I’m completely lost as to what could be the issue.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

Bombastisch, 2018-08-08:
Did you load the keys correctly?

Click on the gear-icon in the topright and it will show you if both keys are loaded.

When does the error appear?
"Failed to **de**crypt " seems kind of a strange error when trying to write, because actually it has to **en**crypt it.

Was this a typo or does it show that error?

Do you own any "real" amiibo? Maybe you can check if you can actually scan them and test if that works.

Please let me know :)

askep3, 2018-08-09:
There were no errors, and it wasn’t a typo. I managed to solve the error by transferring the amiibo bin files from a pc. It has something to do with downloading the files directly from the smartphone. When I’d load the bin files in, the info at the top of screen would only say unknown, but after I used the exact same bin file (except transferred from the pc), the issue was gone. Both the keys were loaded in, btw.

Bombastisch, 2018-08-09:
Okay good to know it works now!

Yes I have heared it a few times before that people had issues with that.
I can't figure out why, but it happend before and now again.

Usually if TagMo does not even show the amiibo correctly in the info-box you can be pretty sure that the .bin file is corrupt **or** the amiibo database wasn't updated yet.