HiddenRamblings/TagMo #137: "Done" although no tag

tobiasmuecksch, 2018-10-30:
I'm using a Samsung Galaxy J5 2017 with Android 7.0. TagMo Version 2.6.1

What I did:

1. Import keys (locked and unfixed)
2. Select an amiibo
3. Press write Tag (Without holding a card to the phone)
4. "Done" is shown immediately

Even if I hold a NTAG215 card to the phone the behavior is the same and the card does not work.

tobiasmuecksch, 2018-10-30:
Somehow the issue disappeared. While "Done" is still shown immediately the write process is working now.

closed 2018-10-30 by tobiasmuecksch

Bombastisch, 2018-10-30:
It's a copy & paste bug in the source code, which was already fixed months ago. We just haven't had a new release yet so the "Done" still shows up.

Don't worry about it, as it actually doesn't mean anything.