HiddenRamblings/TagMo #140: Is TagMo Still being updated?

NotAlbertYee, 2018-11-13:
There are a lot of great features that were being worked on but haven't been released for almost a year. Is TagMo development still in the works or has it ended?

Bombastisch, 2018-11-13:
It's kind of the stagnation we have had a little bit over a year ago (IIRC?).

I'd love to see updates on TagMo again, since with the release of Super Smash Bros Ultimate, there could be a major increase in users and updates might be needed to support that.

It's sad that it stopped like that, I am still hoping for a comeback! :)

jayayou, 2018-11-18:
I genuinely cannot figure out how to install this on my phone. I've downloaded the latest release, and 2.4, but I don't know what to do with code. Without an apk I'm so lost.

closed 2019-06-27 by Bombastisch