HiddenRamblings/TagMo #141: Any update for SSBU?

Ahellina, 2018-12-13:
SSBU amiibo data doesn't work with tagmo

jivandabeast, 2018-12-14:
I hope so, although it's only been a week and I imagine it'll take some time for tagmo to get updated to work properly with ssbu

stilicrafter, 2018-12-15:
That means this App wont work for the switch?
(I just orderd a switch so i couldnt test the app generated tags)

Bombastisch, 2018-12-15:
Just to clear up some facts before wrong information is spread:

TagMo generated amiibo will 100% work with any switch supporting amiibo

TagMo just doesn't support editing the data specifically for SSBU yet. So you can't cheat your level and edit the special attacks.

smokeandlights, 2018-12-27:
I just wanted to say that I also hope that HiddenRamblings will update this excellent app. Thanks for all the work you've done, and keep up the good work!

evinjaff, 2019-01-18:
Yes, hopefully, HiddenRamblings will continue to update this. I'm incredibly grateful for how far this project has come, and regardless of whether or not SSBU editing gets added, I will continue to use TagMo. I also assume that editing might be hard to implement with the spirit system in place.

mictibo, 2019-04-26:
I turned on my NFC data exchange on my phone Lenovo P2a42 with Android v7.0

I also just downloaded the 2.6.1 APK from 
https://nfc-bank.com/bins.php?do=download&downloadid=1801

After installing I downloaded and seletec from TagMo locked-secret.bin and unfixed-info.bin

I then downloaded all content for SSB at
https://nfc-bank.com/bins.php?do=download&downloadid=1683

Then, using TagMo, I can load in the app amibos.

From there, I can't figure out how to load it in my nitendo switch.
Overing the phone over the right handle generates and error from the Switch:
''An error occured while reading amiibo data. Please try again.''
And this with any amiibo that I try.
I also tried whileclicking SCAN TAG, WRITE TAG and RESTORE TAG.
Always the same result.
Thanks for your help. There is probalby something I am doing wrong or not understanding

Bombastisch, 2019-04-26:
You need some NTAG 215 . 
Those are small NFC chips like in amiibo themselves. You first need to write the data to them and then use the NTAG 215 as amiibo.

Read the Readme and Wiki for further instructions.

mictibo, 2019-04-29:
So I already have a power tag that is reprogramable,.
So there is no way to only use the phone.
Thanks for your help

littlehackboi, 2019-10-26:
please this would be so useful with ssbu

nath30, 2020-02-26:
Any chance of it happening now since @Bombastisch updated/ fixed it
there is some kind of editor out there
[GBATemp](https://gbatemp.net/threads/ssbu-amiibo-editor.529499/)
[GitHub](https://github.com/odwdinc/SSBU_Amiibo)

BS13421, 2020-03-10:
It'd be amazing for people who have access to an android but not a windows to be able to edit the ssbu data and not just the sm4sh data. Of course it's still an amazing project and I use it super often, but it'd be great and maybe increase the popularity of things like this with the possibilities of how far the attack and defense can go

AbandonedCart, 2021-09-11:
Closed per the updates mentioned in https://github.com/HiddenRamblings/TagMo/issues/130

closed 2021-09-11 by AbandonedCart