HiddenRamblings/TagMo #145: USB NFC Reader/Writer

UnAfraid, 2019-01-16:
Hello,

Is there any way to use external NFC Reader/Writer with TagMo application?
I have that one https://www.amazon.co.uk/gp/product/B00GYPIZG6

Bombastisch, 2019-01-16:
Well there is this:

https://github.com/socram8888/amiitool

This script runs at the core of TagMo and you can use it manually.

Here is the original thread with the concept:
https://gbatemp.net/threads/diy-amiibo-cards.406978/

(This thread also kickstarted the development of TagMo)

UnAfraid, 2019-01-17:
Is there any tool i could use to write the amiibos to the tags?
I got 10 NTAG215 tags at hand

Bombastisch, 2019-01-17:
Is there not a program shipped with the writer you own?

There should be one according to the details on Amazon. Or atleast there should be some download site mentioned in the manual.

closed 2019-03-17 by Bombastisch