HiddenRamblings/TagMo #147: Sync Amiibo info with AmiiboAPI failed

Chakan516, 2019-03-29:
I keep getting "Syncing Amiibo info with AmiiboAPI failed" when I try to sync the information and would like to see if this is a known issue or just something I'm doing wrong.  Several months ago it worked perfectly but as of today it won't sync anymore.  I'm using the same Galaxy S8+ I was using before.  Any help would be greatly appreciated.

[image]

Bombastisch, 2019-03-29:
The Amiibo API switched to HTTPS a few days ago.
Since TagMo is still trying to get the file via HTTP it is failing.

Thos can only be fixed by @HiddenRambler or @North101 with a new update for TagMo.

Until then you will have to do it manually.
Download the .json file from https://www.amiiboapi.com and use the import option in TagMo to load the file.

Chakan516, 2019-03-29:
Thanks for the reply..  I'll try that in a sec.

D-Pyro, 2019-03-31:
I was trying to recompile with the changes but because It's so out of date I get a bunch of compile errors. So here's a [quick hack](https://drive.google.com/file/d/1WGxZahl2Au0WsmLUhk_wplY-yvCop-ET/view?usp=sharing) to get the existing apk to work.

Chakan516, 2019-04-14:
Sorry it took so long to reply...  Thanks both of you.  Got it working(ish) and setup all of the Amiibos as coin backups.

Rynelan, 2019-04-16:
@D-Pyro thanks for the changed apk! I needed that one 😁

LordHemlock, 2019-04-19:
> I was trying to recompile with the changes but because It's so out of date I get a bunch of compile errors. So here's a [quick hack](https://drive.google.com/file/d/1WGxZahl2Au0WsmLUhk_wplY-yvCop-ET/view?usp=sharing) to get the existing apk to work.

Do you think you could submit the changes you made as a PR?  Would be worth it to have that kind of fix on the master build and not just hidden away in an issue.

D-Pyro, 2019-04-19:
It's a dirty hack of the original apk. All I did was find the url and added an s. I didn't compile the code.

Bombastisch, 2019-04-19:
That's not a dirty hack. Basically that's what is needed to be done. But I can't release a apk with the right key signature because only @HiddenRambler has the key.

mbirth, 2019-05-06:
Does the quick hack also include the AmiiboAPI changes? And @Bombastisch, what about releasing the APK with your sig? It's not a big hassle to uninstall the app first before installing your version. Seems like @HiddenRambler is MIA anyways.

wzhy90, 2019-05-18:
Another dirty hack, need root and dont need to re-install app or uninstall it.
If U R running android 8 or higher, the patch will be ``/data/app/com.hiddenramblings.tagmo-[random-base64-string]/oat/arm/base.vdex``
android 7 or earlier: ``/data/dalvik-cache/arm/data@app@/com.hiddenramblings.tagmo-[x]@base.apk@classes.dex``

run these command in adb shell or terminal app after force stop the app
```
cd /data/app/com.hiddenramblings.tagmo-[random-base64-string]/oat/arm/
sed -i "s|http://www.amiiboapi.com/api/amiibo/|https://www.amiiboapi.com/api/amiibo|g" base.vdex
```
or
```
cd /data/dalvik-cache/arm
sed -i "s|http://www.amiiboapi.com/api/amiibo/|https://www.amiiboapi.com/api/amiibo|g" data@app@/com.hiddenramblings.tagmo-[x]@base.apk@classes.dex
```

After that you can sync online.
tested on android 8

Bombastisch, 2020-01-17:
Fixed in newest release.

closed 2020-01-17 by Bombastisch

Tymoth, 2020-11-29:
Hey all,
I was also trying to get the recent Amiibo JSON file for TagMo since the old file is missing a lot of Amiibos. Of course it is still possible to scan and save them as binary file to write it on new Tags, but as I'm a perfectionistc IT-Pro (🤓) I would like to see all properties correctly for all Amiibos. So I compared the old JSON file in my TagMo-App with the new one from AmiiboAPI and recognized the totally changed format of the file. The new one has more properties that were before integrated in the "Title"-Hex Code of the Amiibo. I'm not really experienced with JSON files so I would like to ask you guys if there is an easy way to bring the new JSON Infos in a format that the TagMo 2.7. version would recognize. I will paste a screenshot from JSONdiff, on the left (old) side there are also lines that look like the first ones for "amiibo_series" that contain "type", "character" and so on:
[image]

Bombastisch, 2020-11-29:
I haven't looked deeper into this, but syncing still works on my phone and I think i also see the correct stats in the amiibo-browser.

Can you elaborate on a certain error you spotted?

Tymoth, 2020-11-30:
I tried to import that new JSON file in the official TagMo 2.7 and I get an error while parsing. So there is no chance to get this JSON in 2.7 and I will have to uninstall it and install the 2.6.1 from the Google Drive link in the thread? Sorry but I'm not really familiar with Github and the procedures.

Bombastisch, 2020-11-30:
I just went into TagMo's in-app settings and hit sync with amiibo API.
Then it displays "syncing was successful" and that's it.
Am I missing something obvious?

Chakan516, 2020-11-30:
The original issue I posted about on 28 March 2019 has been fixed both with the old download provided by the awesome @Bombastisch & the update in Januaryish.  It may be an issue on other phones but it works perfectly on my Galaxy S8+ now.

Tymoth, 2020-12-22:
I got a new phone some days ago (finally Xiaomi Redmi-phones have NFC) and tried again and installed the tagmo 2.6.1 APK but it still can't parse the Amiibo Info with a downloaded .json from amiiboapi and the sync in the app still doesn't work because of the missing https fix. My phone is not rooted, so the method from wzhy90 is also not an option. 
@Bombastisch you posted in January that something was fixed in newest release, but I assume that this is already in the 2.6.1 APK that I have on my disk? Or where can I find this fixed version?