HiddenRamblings/TagMo #149: NTAG512 detection on Xperia phones

NevetsTheSteve, 2019-05-24:
Hello Everyone,
I downloaded TagMo-2.6.1 on my Japanese Xperia phone (Xperia XZ2) and it couldn't read or write tags with an error message of "Error getting tag data. Possibly not a NTAG215".

So I downloaded the 2.6.1 source code and did a little bit of research. 
From what I can tell, TagMo is using the NFC MifareUltralight technology.  On my Xperia phone(s) NTAG512 tags do not support the MifareUltralight technology.  While the NTAG512 chips do support the same interface as Mifare Ultralight and share the same ATQA and SAK values, apparently they are not technically the same.  

I do not know if this lack of support is intentional on Xperia's part however.  I did find a workaround that I wish to share here.  

While the MifareUltralight NFC technology is not supported, the NFCA technology IS supported.  For those that are unaware, NTAG512 and MifareUltralight chips are both based on NFC-A.  I have created a simple compatibility wrapper around the MifareUltralight and NfcA classes. 

The NTAG512 class is constructed similarly to the MifareUltralight class.   Calling NTAG512.get(tag) will internally attempt to get an instance of the MifareUltralight class.  If this succeeds, subsequent calls are routed in to this class instance.  If this fails, it will try to get the NfcA class.  Since the NfcA class is lower level than the MifareUltralight, some functions had to be reimplemented using the tranceive function. Most of these were more or less copied from the MifareUltralight class,  

I have attached the files I modified to this post.  
Changes:
NfcActivity.java :  changed MifareUltralight.get(tag) to NTAG512.get(tag)
TagWriter.java: changed all references of MifareUltralight to NTAG512
NTAG512.java:  created wrapper class to manage differences between MifareUltralight and NfcA classes.

Thank you for reading,
Steve

References:
NTAG512 data sheet:
https://www.nxp.com/docs/en/data-sheet/NTAG213_215_216.pdf

[NTAG512.zip](https://github.com/HiddenRamblings/TagMo/files/3215405/NTAG512.zip)

NevetsTheSteve, 2019-05-24:
I only just now realized that I named the class wrong 512 not 215 ... sorry

kunikunn, 2019-08-29:
I am using Xperia XZ1.What should I do to use TagMo.

Osaka-Taro, 2019-12-25:
Hello NevetsTheSteve
I built it under the following environment, but I got an error.
Android Studio 3.5. 3
Java SE 8u 231

Run: gradlew.bat
Error: NullPointerException

In what kind of environment did you build?
If you like, please upload apk.

NevetsTheSteve, 2020-01-14:
I have create a fork and pull request to add this compatibility fix and make it compile with the Android Studio 3.5.3

Osaka-Taro, 2020-01-15:
Dear.
I was able to compile TagMo based on the fork you made and CMake information from brunomarchand.
Also, I was able to write properly on the Android device that had a write error.
Thanks for the great changes.

Bombastisch, 2020-01-17:
Implemented in newest release.

closed 2020-01-17 by Bombastisch