HiddenRamblings/TagMo #150: Stuck at done on pixel XL

arkhamsaiyan, 2019-05-28:
When scanning NFC tag on rooted and non-rooted pixel XL tags properly loaded but screen still doesn't flash back to Maine when trying to scan or save new NFC tag

Bombastisch, 2019-06-24:
Does this occur on every release of the app?

terryskorski, 2019-07-08:
I didn't see the OP provide any update, but I am having the same issue. I have a Pixel 3a XL, and when you hit Write the bottom instantly shows "done". After about 3 seconds the 'done' disappears and then the app just hangs there. I have tried 2.6.1, 2.3.4Beta, and 2.3.1. The last of which errored out on the locked-secret.bin and unfixed-info.bin. Any thing I should try to get this to behave normally?

Bombastisch, 2019-07-10:
So the "done" that shows up right in the second you press "write" is just a small bug which was fixed since the last release ( https://github.com/HiddenRamblings/TagMo/commit/01edf4ba78140098e4446875742c800c8ca75f13 ) 

It actually doesn't mean anything.

Does it write the tag? Or are you sure you put the NTAG 215s  right over your NFC-writer of your phone? Maybe try put your phone case off.

arkhamsaiyan, 2019-07-11:
So actually what had happened I'm sorry I didn't get back to this I had no idea that you had responded or that there was any activity on it and I apologize. I don't know exactly what had happened but I do know once I switched my phone from a rooted pixel XL that had the back replace because the glass on the pixel XL is the piece of crap... As soon as I upgraded from the pixel XL to the S10 plus I was able to scan and write tags flawlessly so if you are not getting a feedback vibration and you have indeed checked that NFC is on, then you're NFC module or chip built in to the phone may be damaged unfortunately.

closed 2019-07-11 by arkhamsaiyan