HiddenRamblings/TagMo #153: App crashes when attempting to load tags

mmmeeggss, 2019-07-08:
App crashes whenever I hit "load tag". Sometimes I manage to open the drop-down menu to choose a folder to search in, usually I can't make it that far. Using a Samsung S10+ on latest version of Android. 

[Logcat ](https://drive.google.com/file/d/1-CTHYB-VDqlAe3JdcNLUjxznGetoS-WC/view?usp=drivesdk)

Bombastisch, 2019-07-10:
You can try to disable the Amiibo File Browser in the settings. Maybe that helps?

How many GBs is your storage? Maybe it's "too" much and so it crashes.

xJam-es, 2019-08-28:
I have the same issue - it works once i disable Amiibo File Browser.

I have all my tags in /TagMo/Amiibo/ and then folders for each category.

I have 37GB free of 118GB (Pixel 2 XL 128GB version).

AbandonedCart, 2021-09-11:
This should be solved by the new storage management methods. Please feel free to stop back if the problem wasn't solved to open this issue again.

closed 2021-09-11 by AbandonedCart