HiddenRamblings/TagMo #154: Android 9/Galaxy A40 compatibility

BallisticBanker, 2019-07-10:
I am currently attempting to set up tagmo on a galaxy A40 running android pie but tagmo does not seem to be reading the SD card as in the browser it only shows internal memory and also has not created a folder. Another issue I have faced is syncing the amiibo api as it pops up with a failed message whenever I attempt to sync. I do have a older phone with android 8 and this phone works fine so I'm not sure if it's the OS

Bombastisch, 2019-07-10:
You can disable the Amiibo File Browser in Tagmo's settings so maybe that helps?

It could be that the OS version is too modern and you need to compile the app yourself to fix it.

I don't have an the problem with your SD card. I have never heard of that before. Might be worth looking into that at some point.

BallisticBanker, 2019-07-11:
@bombastich
Hey it seems that tagmo no longer requires it's own folder in the sd as it can read anywhere on the internal memory. The text on amiibo file browser made it seem as though the app would make a dedicated folder on the sd. 
As for the API according to people on reddit  something updated on the API and tagmo can no longer access it. Right now I've imported it manually from the amiiboapi.com github page

closed 2019-07-11 by Bombastisch