HiddenRamblings/TagMo #162: Transceive fail tagmo 2.7.0

julia12321, 2020-03-13:
Hi, I am trying to use tagmo 2.7.0 to write on a new NTAG215 with a samsung galaxy s4 but the transceive fail window keeps popping up. Is there a way to fix this? Thank you

Bombastisch, 2020-03-14:
Turn off tag validation in the settings. Samsung phones can sometimes be picky.

pemsl2, 2020-04-12:
is there another solution?
i have the same error and turning off the validation didn't fix it
any and all help is appreciated

Bombastisch, 2020-04-12:
Do you have a phone case?
Remove it and try again.

Do other NFC apps work? Can TagMo read amiibo?

pemsl2, 2020-04-13:
thanks for the response.

no i removed the case before attempting the transfer
yes while reading other nfc apps and tagmo work

my only guess right now is, that the cards i use are not compatibel, i ordered different ones and will try with them again

AbandonedCart, 2021-09-11:
This issue appears resolved, but please let us know if new cards were not the solution.

closed 2021-09-11 by AbandonedCart