HiddenRamblings/TagMo #164: Failed to Decrypt Tag

JustOkayAnimations, 2020-03-27:
I can load the tag fine, I can load other Amiibos fine, but when I try to write to my sticker, it says failed to decrypt tag, every time! I'm using a downloaded .bin file, could that be the problem? I'm using NXP NTAG215 stickers if that helps.

JustOkayAnimations, 2020-03-27:
I have tried using the .bin file downloaded from both my phone and computer and when I try to write to the tag or restore tag (with allow restore to different tag enabled), it still fails to decrypt the file. Both unfixed key and locked key say "OK." in the menu.

Bombastisch, 2020-03-27:
Do not use tags that you don't own.

Does it really say "decrypt" ?

If you are actually writing a tag it should say "encrypt". Are you sure you clicked on the correct button, namely "write tag"?

Malicious .bin could also be a problem.

JustOkayAnimations, 2020-03-27:
I haven't seen any text that says "encrypt". I scan the sticker, it says "<Blank Tag>". I tap "Write Tag", scan the sticker again, and it just shows "Failed to decrypt tag".

JustOkayAnimations, 2020-03-27:
I just watched [this video](https://www.youtube.com/watch?v=7oalLcHy1Tk) and i didnt see him mention my issue. Maybe it is the .bin?

JustOkayAnimations, 2020-03-27:
Tried a different .bin with a different animal crossing amiibo from a different series. Same issue.

Bombastisch, 2020-03-27:
Have you tried cloning an existing amiibo? I can't see the problem rn. It seems like bad .bins.

JustOkayAnimations, 2020-03-27:
I tried writing to the sticker without an amiibo .bin on my phone. Still failed to decrypt tag. It's the sticker isn't it? [These are the ones I bought,](https://www.amazon.com/TimesKey-Stickers-Compatible-Nintendo-Programmable/dp/B075CFXY8V/ref=sr_1_5?dchild=1&keywords=ntag215+stickers&qid=1585331252&sr=8-5), are these not compatible or did I get faulty ones?

Bombastisch, 2020-03-27:
Could use the "NXP tag info" app from the playstore and send screenshots, so we can verify?
The link on amazon seems to sell the correct Tag type.

Bombastisch, 2020-03-27:
Do you have any original amiibo to scan?

JustOkayAnimations, 2020-03-27:
Yes, I tried one of them and it worked perfectly fine. [This is the full scan info.](https://pastebin.com/WEc20jxv)

Bombastisch, 2020-03-27:
Seems like you got faulty ones...

Look at the full scan data of NXP tag data.
All the data pages are marked with "+r".
The bottom description says they are writeprotected.

You might want to check if they are all write-protected or if only some of them are faulty.

JustOkayAnimations, 2020-03-27:
ok, ill scan them all and get back to you, thanks

Bombastisch, 2020-03-27:
So i just looked further into this.
It seems like with the app "nxp tag _writer_ " you might be able to remove the write-protection, since the bytes are not locked, but only blocked. Might work.

Otherwise i don't see a lot of chances

JustOkayAnimations, 2020-03-27:
Downloading the writer app right now. I scanned two or three of the other stickers and there were dots where the +r or *r's were, but a +P on the last two lines.

JustOkayAnimations, 2020-03-27:
I tried unprotecting and erasing the tag with the "+r" via the writer app and it says that the tag cannot be unprotected.

Bombastisch, 2020-03-27:
So I just scanned one of my empty tags, and the result looks diffrent.
https://pastebin.com/44UBDsuG

Your empty tags have write protection on, like only amiibo after they were made should have.

You can still try and check all of your tags, but it seems like they are faulty, since they were locked before shipping.

Sorry, seems like the only option seems to return them and order new ones from a diffrent seller.

JustOkayAnimations, 2020-03-27:
Should I try to contact the same seller first? I know that the stickers are cheap but I'd like to save as much money as possible.

Bombastisch, 2020-03-27:
You usually can get your money back on amazon orders.
Just login, go to previous purchases, select your tags and click that you would like to return them.
Should get all your money back and can buy from a diffrent seller.

Good luck! Sometimes it can be a bit of a problem, but usually you get working ones.

Or try to contact the seller and ask for new ones and explain to them why it didn't work.

JustOkayAnimations, 2020-03-27:
Will do, thanks.

closed 2020-03-27 by JustOkayAnimations