HiddenRamblings/TagMo #166: Tag keys won't load nor write. Keep getting errors.

ChaosQueen713, 2020-03-28:
I had the older version and  I had the BIN files for the amiibo cards I wanted to make and it has worked before. I load the Tag and when I hit Write NFC and put the nfc card under it I get the Source UID does not match the target. I have used these before with different characters from another series and it worked perfectly. 

I updated the app to the new version and load the Amiibo key I wanted but Write tag was greyed out and it kept saying Amiibo Key not loaded. I go into settings and don't exactly know what to click so I uninstalled and Deleted everything.

What do I need to do to get these working? It scans the blank cards just fine. I do have the Samsung S9+ phone now while before I used the S7+ and prior models.

Bombastisch, 2020-03-28:
You cannot overwrite amiibo cards.

Once they are an amiibo you can't revert it or change the amiibo type.

To load the keys, open the settings and click on the red error "keys not loaded" to open a file dialog and load the keys.

ChaosQueen713, 2020-03-28:
I am wanting to make a card. I have blank NFC tags. I do hit the triple
dots and find a load a Tag, yet I still get the error message that one is
not loaded. The Write button is greyed on the new version so I cant. On the
older tagmo version when i hit write it scanned the blank NFC tag but gave
me the Error Code Source UID not found.

Bombastisch, 2020-03-28:
You don't load a .bin via the three dot menu.

Just click on the violett "load tag" button and select a bin from there.

ChaosQueen713, 2020-03-28:
Sorry I got mixed up there.
But yeah I did load the .bin file through the load tag button and I can't
Write the tag to a blank NFC. The Write button is greyed out

Bombastisch, 2020-03-28:
It's usually only greyed out when NFC is turned off or the keys aren't loaded.

Have you loaded the keys?

ChaosQueen713, 2020-03-28:
At the time yes.
I think I'll download it again and redownload all the keys I had and see if
it works.

ChaosQueen713, 2020-03-28:
> It's usually only greyed out when NFC is turned off or the keys aren't loaded.
> 
> Have you loaded the keys?

I took screen shots of what I see and do on my phone. What can I do?

I keep getting this https://imgur.com/a/ri2i0Im

Bombastisch, 2020-03-28:
It clearly shows a red error in the settings which I mentioned before.

https://i.imgur.com/9xncXlI_d.jpg

The keys are not loaded. Click on that error message and load the key(s).  It's usually a 40byte .bin file

ChaosQueen713, 2020-03-28:
Where would I find they keys? Are they part of the tags download? I'm not
finding the bin file.

Bombastisch, 2020-03-28:
Please read the Github readme -> requirements.

I cannot provide those.
Googling can help.

ChaosQueen713, 2020-03-28:
Ah I thought the load keys were something different that could be. My bad.
Thank you!

Bombastisch, 2020-03-28:
Does it work now?
Can I close the issue?

ChaosQueen713, 2020-03-28:
Yes please! It.works.now thank you!

closed 2020-03-28 by Bombastisch