HiddenRamblings/TagMo #169: Samsung Galaxy S4 wont allow me to install the apk at all.

Sejha, 2020-03-30:
Title

Sejha, 2020-03-30:
To clarify, actually, going to the install page for the apk just results in the only option being cancel.

Bombastisch, 2020-03-30:
You need to enable "install app from unknown sources" in your phones security settings to install TagMo, as it's not available on Google Play

Sejha, 2020-03-30:
I am well aware of this and have the setting turned on, I simply am somehow not allowed to click install on the page. I mean its kind of obvious its not on play when Im downloading the APK directly, Ive done this before, its just THIS app that wont load.

Bombastisch, 2020-03-30:
Which Android version are is your phone on.

In the newer versions (<7.0 ?) you need to grant the permission per app. So maybe you need to give the unknown sources permission to your file explorer.

Other case could be that  you still have the older version installed.

Please uninstall it beforehand. I already put a notice in the download page.

Sejha, 2020-03-30:
Ah that might be it, I'll see if I can change the permissions on the actual file explorer as Im using the latest version of android available to my phone.

Sejha, 2020-03-30:
Alright just went in and checked and Im on android 5.0.1, model SGH I337-M

Bombastisch, 2020-03-30:
Not sure when Google exactly changed that. Was just a guess.

Does it work now?

Sejha, 2020-03-30:
I actually for some reason am not allowed to change the permissions for the my files app, and feel like there's some sort of cascade of options preventing me from installing it. I'll do some more research but this is a pain.

Bombastisch, 2020-03-30:
One option would be to give the permisson to chrome/firefox, download TagMo again and then install the app right from the push notification.

Another option would be to give that permission to the android integrated "downloads" app and install from there.

Sejha, 2020-03-30:
Well Im fucked, apparently I can't do this without rooting the device which Im not willing to at this time. This is beyond frustrating as it used to be easy to do it.
Why is *this* apk the only one that it doesnt work with.

Bombastisch, 2020-03-30:
TagMo still does not need your phone to be rooted.

I use it on my unrooted phone as well.

- Have you tried using an older version of TagMo?
- Did you check if you still have an older version of TagMo installed? You need to uninstall an older version first to install 2.7.0.

I can't replicate your problem, as TagMo is straight forward to install, your phones just needs the permission.

Sejha, 2020-03-30:
Changing the permission is what is what I need to root for as there is no option available on my version of android and Id have to downgrade to get the option back. I have no idea why this is the way it is.

Bombastisch, 2020-03-30:
There are lot's of people running TagMo on a S4, i don't think you need root for it.

If you want to put further effort into this feel free to send screenshots into this thread of your phone not wanting to install TagMo.

eG. send me a screenshot of your sertting's screen showing that permission to installing unknown apps is actually set for your browser

Sejha, 2020-03-30:
I think theres a critical reading failure going on here. I cannot set that permission. It won't allow me to access that as an option.

Bombastisch, 2020-03-30:
Huh... Okay thank you for clarification.

That's still odd.
Just one more try, since all other S4s seem to be able to run TagMo.

Does your settings app have a searchbar?
Type in "unknown".
Usually you can find the correct setting really quickly that way, as that option is buried somewhere deep in the settings.

Sejha, 2020-03-30:
It does not have a searchbar at all, Unknown Sources only shows up once anywhere and its in security and its been turned on now for a while, I even flicked it off and on again and restarted the phone to see if it was a stuck setting. Still no dice.Other nfc programs work, even the old Unsafe Amiiwrite from years ago but no such luck here. I can't quite figure it out.

Bombastisch, 2020-03-31:
I sadly don't know either then...

Can you install other .apk from unknown sources?
e. g. download and install F-Droid

Sejha, 2020-03-31:
Yeah theres no issue with other APK it seems its just this one that it wont let me do and I cant put my finger on it. I'll get back to you if I come up with anything.

Sejha, 2020-04-07:
FIGURED IT OUT AND I AM FUCKING LIVID
I was using twilight as a screen filter and apparently as a system level protection it will not allow you to click install when its running. I have no idea why it was only THIS app that was doing it but hey I got it fixed.

closed 2020-04-08 by Bombastisch