HiddenRamblings/TagMo #172: Lost tag and scan issue

luellessi, 2020-04-02:
I've been trying to write to a NFC215 tag using the newest version (and the one before that) and have not been successful. I have Samsung Galaxy S9 with Android 10.

For most of the time the app keeps endlessly scanning for a tag, sometimes I get 'tag detected' only to be quickly replaced with 'tag lost' message. Sometimes I also get 'failed to transceive', and 'error while writing main data (stage 1) android.nfc.taglostexception: tag was lost'. Mainly though it keeps endlessly scanning.
Thinking I got invalid sticker tags I tried scanning an official amiibo card - but it just keeps scanning.

I tried disabling NTAG215 validation and also charging the phone fully and connecting it to the PC which seemed to worked for someone here, but it doesn't for me.

Any ideas what I might try to get it to work?

Bombastisch, 2020-04-02:
Do you have a case on your phone? Try without it, sometimes the case is to thick to allow proper NFC

luellessi, 2020-04-02:
Unfortunately I've been having the problem with and without the phone case:(

Bombastisch, 2020-04-02:
So scanning does not work either, right?

Just to get the basics, have you tried reebooting your phone?

Sometimes phones, mine included are very picky abt getting the right contact to NFC chips. Look up a diagram online of your phone to see where exactly your NFC sensor is and hold a real amiibo there.

Try to not shake your phone or the amiibo it interrupts reading/writing very easily.

Have you tried using an app like "nxp tag info" from the playstore to see if your NTAG215s are working?
Does that app have NFC communication problems as well?

luellessi, 2020-04-03:
I installed the nxp tag info and tried scanning my official amiibo card, but it took forever to finally detect it, but it finally did, same goes for the tags I bought. I keep getting 'communication lost messages while scanning though but it scans them fully I think. 

Meanwhile I just left official amiibo under the phone and started scanning and after forever it scanned it.... so I tried writing a tag... and also after many 'tag lost' and 'error while writing' it did finish writing!:)

So... I guess it works but once in a blue moon. Is there any way to improve detecting tags by my phone? Like an app or something?

At least I confirmed that it is not Tagmo or the tags problem, maybe just my phone.

Bombastisch, 2020-04-03:
People here: https://forum.xda-developers.com/galaxy-s9/help/nfc-problem-t3786656

complain about NFC problems too. Apparently needing more than 73% charged battery to have NFC working.

If you still have warrsnty on your phone you can probably send it to Samsung and ask for repair, but I am not sure if that currently works with Covid19.

Since other NFC apps have similar problems, it's safe to say it's a hardware problem and can't be fixed through software.

luellessi, 2020-04-03:
Yes, I thought of doing that after the situation gets better.
Thank you for the help!

closed 2020-04-03 by Bombastisch