HiddenRamblings/TagMo #181: Transceive failed

maddsn, 2020-04-18:
I try to write my nfc tags with the newest version, but it doesn‘t work and I don‘t know why. I tried different data files, but that didn‘t solve the problem. Does anyone have any tips to solve this problem?

Eldin-Bridge, 2020-04-18:
Can't detect a file when you open the app? or it marks you error when writing the data in the blank card?

Bombastisch, 2020-04-19:
Are you sure your tags work?

Bombastisch, 2020-04-19:
Use the "NXP taginfo app" from the playstore and send a screenshot of the full scan of one of your empty tags.

Tomatenjoghurt, 2020-05-22:
Hi, having the same problem. Tried it with my Pixel 3 and OnePlus 7 of my Girlfriend. It was working some weeks before but all of a sudden it stopped working.
Having Tags from the same vendor here which I wrote sucessfully some weeks ago.
TagMo says when trying to scan the empty tags "Transceive failed". Real amiibos are read perfectly fine.
NXP Tagwriter recognizes the tags as NFC215 Empty Tags. I've attached the TagMo log dump, it looks like there is a file missing?
Maybe an issue with new Android 10? IDK...
[tagmo_logcat.txt](https://github.com/HiddenRamblings/TagMo/files/4668509/tagmo_logcat.txt)

Bombastisch, 2020-05-22:
Scan them in NXP taginfo and check if the bytes in the "fullscan tab" show up as not locked.

Tomatenjoghurt, 2020-05-23:
Nevermind, I grabbed the wrong NFC stickers from the shelf. I tried to use NFC 203 stickers, thats why I received that error message.
Feeling so stupid right now, thank you!

closed 2020-05-23 by Bombastisch