HiddenRamblings/TagMo #182: File Browser Failed

Eldin-Bridge, 2020-04-18:
When I installed the app it worked for me a few days, but then it started to close every time I loaded the tags, so I decided to deactivate the file search and it worked, I think it needs a new update focused on this problem, I have a samsung galaxy A8, I would like to contribute in some way to be able to fix it and launch version 2.7.1 correcting this error, in addition to this I have had the problem that the nintendo switch does not detect the card as an amiibo despite having been written correctly, it has only happened to me with 2 data that I throw an error.

Bombastisch, 2020-04-19:
Can you use Android Studio?
Compile the master2 branch and run it. Maybe switching to AndroidX helps and fixes it

Eldin-Bridge, 2020-04-20:
I haven't used android studio but I have knowledge of arduino and PLC programming, I will try to work on this app to improve it by adding functions and correcting errors, first I will try to see what the android studio app is about and then I get to work, I suppose that the app It is based on Java, right?

minalou, 2020-04-22:
WHY do you send your mail to me?? Fuck...

Eldin-Bridge, 2020-04-22:
Hi, I don't remember sending you my email 🤔 but I apologize for any misunderstanding.

AbandonedCart, 2021-09-11:
The methods discussed in this issue are no longer a part of the core functionality and have been completely rewritten.

closed 2021-09-11 by AbandonedCart