HiddenRamblings/TagMo #183: master2 branch App crash due to Collections.sort

zxavier94, 2020-04-22:
com.hiddenramblings.tagmo.BrowserAmiibosAdapter.AmiiboFilter#publishResults  Collections.sort((ArrayList<AmiiboFile>) filterResults.values, new AmiiboComparator()); 
this line code cause the App crash

zxavier94, 2020-04-22:
android phone: redmi note 8 pro
os: android 9

Eldin-Bridge, 2020-04-22:
In which program did you check it?

zxavier94, 2020-04-23:
> In which program did you check it?
I download release 2.7.0 and install on my phone.When I open the app and search amiibo bin file,the app always crash.so, I pull the master2 branch to local，and use android studio build, and debug the app.Finally,I find the problem code.After I comment the code , and build the app, the app can run normally.

Eldin-Bridge, 2020-04-23:
Excellent, if you can support that this program can improve, contact the manager to upload a review of the problems that you correct, I haven't had time for now but when I can I will try to help, all good intention thanks, if it doesn't bother you and you want to support non-profit go ahead

zxavier94, 2020-04-23:
> Excellent, if you can support that this program can improve, contact the manager to upload a review of the problems that you correct, I haven't had time for now but when I can I will try to help, all good intention thanks, if it doesn't bother you and you want to support non-profit go ahead

Thank you!I am a gamer.These days i like Animal crossing,so I'll be glad to help others.I will try to  fix the problem tomorrow or on this saturday,then i will contact the manager and push the code.

zxavier94, 2020-04-24:
Sorry,I have token 2 hours to try to fix the problem,but i failed.I change the way of reading the Amiibo.json file.
[image]
I let the app read Amiibo,json from External Files Dir,because the app can't read file from the root path,then i comment the Collections.sort code.
[image]
Finally,the app can run normally.I don't know why,the com.hiddenramblings.tagmo.BrowserActivity file is really complicated.

AbandonedCart, 2021-09-11:
This should be resolved in the newest versions, but please feel free to comment back if this issue needs to be opened again.

closed 2021-09-11 by AbandonedCart