HiddenRamblings/TagMo #189: Amiibos can no longer be deleted from the cards.

Luca-g14, 2020-04-27:
I go on allow that I can delete the amiibos from the cards.  But it just doesn't work.  😥 When I use the function on the card it says that it is done, but that is not true. The cards are still occupied with the previous amiibo.

critterm, 2020-04-27:
This is happening for me too

Bombastisch, 2020-04-27:
You cannot delete amiibo once they are written onto NTag215s.
Just like real amiibo they are locked once they are written.
The Readme clearly states that, but I think many just overlook that.

If you want another amiibo you need to use another tag.

reactor-sa, 2020-04-27:
NTAG215 is not re-writable, nothing to do with the software

closed 2020-04-27 by Bombastisch

Luca-g14, 2020-05-03:
but I have already deleted a card and put a new amiibo on it