HiddenRamblings/TagMo #197: When you click load tag it just crashes and goes back to the front screen

lizziem25, 2020-05-16:
Please can someone help me ASAP every time I click load tag it just crashes and goes back to the front screen. Tried rebooting the phone

LeVraiRoiDHyrule, 2020-05-22:
I've got the same issue here. Redmi Note 8T with Havoc OS 3.5, Tagmo latest version installed. When trying to load bin tags, Tagmo just crashes.

Bombastisch, 2020-05-22:
Probably related to #183.

I will look into it... can't promise to fix it soon

reylas81, 2020-05-28:
I am having the same issue since yesterday.

I upgraded my phone from a Samsung S9+ to the new S20+ 5G... it worked until I had a System Update this morning...

skizdidlyidler, 2020-05-30:
This started happening to me after adding the SSB Ultimate amiibo folder last fall (2019). I was still using an older version at the time which prompted me to look for an update. But the newer 2.6.1 was also doing the same thing. The only fix I have found is to disable the internal file viewer "Enable Amiibo File Browser" in the settings, which is very unfortunate, as I much prefer the UI of the Amiibo File Browser.

reylas81, 2020-06-01:
Disabling the TagMo amiibo browser fixed the issue for me!

closed 2020-12-22 by Bombastisch