HiddenRamblings/TagMo #200: Error while writkng main data

Oreosmlm16, 2020-05-30:
I get thr java error while writing tags. I made 10 normally and they worked just fine. Then my phone updated,  and now I get this error every time. The tags I bought were from the reddit guy so he said they are verified working. And the taginfo app appears that are correct. I have no idea what is going wrong. Help  please! I was making these for a child's bday party.

Bombastisch, 2020-05-30:
Can you send a screenshot of the full scan in taginfo?

Oreosmlm16, 2020-05-30:
** TagInfo scan (version 4.24.6) 2020-05-30 10:27:11 **
Report Type: External

-- IC INFO ------------------------------

# IC manufacturer:
NXP Semiconductors

# IC type:
NTAG215

-- NDEF ------------------------------

# No NDEF data storage populated:

-- EXTRA ------------------------------

# Memory size:
504 bytes user memory
* 126 pages, with 4 bytes per page

# IC detailed information:
Full product name: NT2H1511G0DUx
Capacitance: 50 pF

# Version information:
Vendor ID: NXP
Type: NTAG
Subtype: 50 pF
Major version: 1
Minor version: V0
Storage size: 504 bytes
Protocol: ISO/IEC 14443-3

# Configuration information:
ASCII mirror disabled
NFC counter: disabled
No limit on wrong password attempts
Strong load modulation enabled

# Originality check:
Signature verified with NXP public key

-- FULL SCAN ------------------------------

# Technologies supported:
ISO/IEC 14443-3 (Type A) compatible
ISO/IEC 14443-2 (Type A) compatible

# Android technology information:
Tag description:
* TAG: Tech [android.nfc.tech.NfcA, android.nfc.tech.MifareUltralight, android.nfc.tech.NdefFormatable]
* Maximum transceive length: 253 bytes
* Default maximum transceive time-out: 618 ms

# Detailed protocol information:
ID: 04:99:80:CA:E8:6B:80
ATQA: 0x4400
SAK: 0x00

# Memory content:
[00] *  04:99:80 95 (UID0-UID2, BCC0)
[01] *  CA:E8:6B:80 (UID3-UID6)
[02] .  C9 48 00 00 (BCC1, INT, LOCK0-LOCK1)
[03] .  F1:10:FF:EE (OTP0-OTP3)
[04] .  A5 00 00 00 |....|
[05] .  FA CE 25 67 |..%g|
[06] .  10 38 F4 BF |.8..|
[07] .  6B 2A 86 F3 |k*..|
[08] .  D1 38 91 9B |.8..|
[09] .  28 78 C6 BF |(x..|
[0A] .  00 C8 0B EE |....|
[0B] .  92 D3 C1 ED |....|
[0C] .  FE AE B0 18 |....|
[0D] .  0F C2 4E 96 |..N.|
[0E] .  19 94 81 45 |...E|
[0F] .  44 AA 15 2B |D..+|
[10] ?  60 1E 5C 3B |`.\;|
[11] ?  47 7F 84 CB |G...|
[12] ?  5F 02 3C 70 |_.<p|
[13] ?  F7 FD 7A 6C |..zl|
[14] ?  11 11 49 16 |..I.|
[15] ?  03 AD 00 01 |....|
[16] ?  01 B2 05 02 |....|
[17] ?  33 36 37 32 |3672|
[18] ?  EC 95 DE 51 |...Q|
[19] ?  CE 64 DA B1 |.d..|
[1A] ?  5C 17 A9 48 |\..H|
[1B] ?  D1 2B 68 39 |.+h9|
[1C] ?  AF 77 5A 90 |.wZ.|
[1D] ?  78 53 FD 92 |xS..|
[1E] ?  71 07 08 21 |q..!|
[1F] ?  84 31 B6 3A |.1.:|
[20] ?  BC CD 41 47 |..AG|
[21] ?  D7 E1 04 46 |...F|
[22] ?  FF 27 26 3D |.'&=|
[23] ?  2E 82 D9 06 |....|
[24] ?  8A 8F F9 2B |...+|
[25] ?  0B 02 43 2C |..C,|
[26] ?  C5 C6 AB A2 |....|
[27] ?  08 1A 79 22 |..y"|
[28] ?  57 4C 26 C1 |WL&.|
[29] ?  42 3B 64 B8 |B;d.|
[2A] ?  C9 75 D3 50 |.u.P|
[2B] ?  3F 4C 7A 89 |?Lz.|
[2C] ?  61 EB A2 F5 |a...|
[2D] ?  01 92 05 E8 |....|
[2E] ?  75 CD 15 AD |u...|
[2F] ?  74 9E 6D 4A |t.mJ|
[30] ?  07 18 5B F0 |..[.|
[31] ?  76 AF C3 9C |v...|
[32] ?  55 8C EC 57 |U..W|
[33] ?  92 E7 50 E7 |..P.|
[34] ?  1C 06 97 86 |....|
[35] ?  41 EE F0 60 |A..`|
[36] ?  98 FC DC DA |....|
[37] ?  37 C9 7F 4F |7..O|
[38] ?  31 95 F8 65 |1..e|
[39] ?  DE F0 8F FA |....|
[3A] ?  66 92 FC C5 |f...|
[3B] ?  00 00 00 00 |....|

<aborted: tag disappeared on page 60>

--------------------------------------

Oreosmlm16, 2020-05-30:
I am hoping this is what you were needing. If not i will try something else

Bombastisch, 2020-05-30:
That's not enough. I need a screenshot of the last tab called "full scan" in nxp taginfo

Oreosmlm16, 2020-05-30:
That is literally all that shows up in full screen.. it always says the tag disappears on page x

Oreosmlm16, 2020-05-30:
[04-99-80-CA-E8-6B-80_2020-05-30 13-51-18_taginfo_scan.txt](https://github.com/HiddenRamblings/TagMo/files/4706251/04-99-80-CA-E8-6B-80_2020-05-30.13-51-18_taginfo_scan.txt)

Oreosmlm16, 2020-05-30:
[04-DB-3D-6A-90-6B-80_2020-05-30 13-53-35_taginfo_scan.txt](https://github.com/HiddenRamblings/TagMo/files/4706252/04-DB-3D-6A-90-6B-80_2020-05-30.13-53-35_taginfo_scan.txt)

Oreosmlm16, 2020-05-30:
[image]

Bombastisch, 2020-05-30:
Look at the last portion of the memory readout. It should mostly be dots on the left side of the grid, indicating that those are unlocked bytes.

But the read fails and shows question marks and then errors completely.
Seems like this tag is broken.
You can try and scan all your remaining tags in taginfo to see if those work

Oreosmlm16, 2020-05-30:
And then the bottom 
[image]

Oreosmlm16, 2020-05-30:
Well these are tags that were bought from ambriios website. And I would think he wouldnt be selling tags that dont work.

Bombastisch, 2020-05-30:
He probably just resells tags from other selleres on ebay/amazon/alibaba.

It can happen to get faulty tags.
I explained to you the reasons why it won't work. 10 other tags you got worked.
Check the rest of your tags in taginfo. It should indicated all bytes as "."  if it is something else it's most likely broken

Oreosmlm16, 2020-05-30:
But after scanning all 30 they all show say the same thing. They say the tag disappears on x page

Bombastisch, 2020-05-30:
Scan one of your functioning amiibo please an send me screenshots of the full scan.

Oreosmlm16, 2020-05-30:
[image]

Oreosmlm16, 2020-05-30:
Which is weird.. considering that scans perfectly on my switch as bluebeard ambiio

Oreosmlm16, 2020-05-30:
And when i scan a legitimate ambiio card, it gives the same error as bluebears

Bombastisch, 2020-05-30:
Do you have a phone case? Try removing it and scanning again. Sometimes it blocks stuff.

Oreosmlm16, 2020-05-30:
After taking the case off the unwritten tag says this 
[image]

Oreosmlm16, 2020-05-30:
And written tag.. says this 
[image]

Bombastisch, 2020-05-30:
The unwritten tag now looks like a blank tag should look like! You should be able to write to it.

Leave your phone case off while using TagMo.
Dunno about the written tag. Maybe it's just a taginfo bug.
I don't have acces to an amiibo right now to test myself.

Oreosmlm16, 2020-05-30:
Thank you for your help!

Oreosmlm16, 2020-05-30:
Yeah even after trying to write the one you said was a clear tag, it still gives the same error. So I think I will hang up my time of trying to do my own. Lol thank you for your help

Bombastisch, 2020-05-30:
Oh shoot... I am sorry. Don't know any further...

closed 2020-05-30 by Oreosmlm16