HiddenRamblings/TagMo #204: Corrupt Tag

RatKing420, 2020-06-03:
Today I got an NFC tag from Tagstand. I was excited to use it to get Stitches in Animal Crossing, one of my favorite villagers. When I scanned the tag, I got an error message saying the tag wasn't valid. I also had the "Failed to Decrypt" issue, but I already tried to fix it. I assume something is wrong with the data on the card. I used a tag reader app and, sure enough, it said the tag was faulty. I already talked to their customer service. Is there anything I can do to fix the card I already have. I have Powersaves but don't have an official amiibo or the tag it came with. Thanks!

Bombastisch, 2020-12-22:
Probably not... For cloning amiibo you need absolutely blank nfc tags, with not data what so ever written on it previously. If bytes have been pre-locked from the seller, they might not work in TagMo.

AbandonedCart, 2021-09-11:
Closing this as outdated, but please leave a comment if the issue needs to be opened again.

closed 2021-09-11 by AbandonedCart