HiddenRamblings/TagMo #205: App crashes when in attempt to load tag.

gengargaryl, 2020-06-05:
Hey there, I had this working just fine with no issues, then I uploaded some more amiibo data and the app started to crash when I wanted to load their tags. I couldn't figure it out, so I uninstalled everything off of my phone and started over with the latest version.

However, it just kept happening. Now, I cannot seem to load any tag without the app crashing completely. Any suggestions out there?

paddykontschak, 2020-06-13:
There might be a specific amiibo data that causes the crash. I have the same issue with Splatoon 2's Marina.

Try either removing Marina from your collection or remove them all and add 1 amiibo after another to find the problematic one.

GewoonIraj, 2020-08-21:
To fix this go to Settings in the TagMo App, then scroll down to Reset Amiibo Info - after it has been reset, hit Sync Amiibo info with AmiiboAPI: it should now no longer crash! Just did it myself and seems to be fixed now.

closed 2020-12-22 by Bombastisch