HiddenRamblings/TagMo #211: Deleting Tags

WeeWigglyWoo, 2020-07-20:
There should be an option to delete tags because I accidentally downloaded loads of duplicate tags that are mixed with the amiibos I already had that were the same character. I tried deleting them through my files but I couldn't and trust me, I tried for almost a solid hour. Until there's an option to delete tags, TagMo is almost useless to me now, destroying any chance of me editing amiibos and getting new ones.

omfgnuts, 2020-07-27:
bruh, just reinstall the app

Bombastisch, 2020-07-27:
Deleting them through your file manager should work. I don't know your exact problem. You can always still reinstall the app or clear the cache if that's an issue.

closed 2020-12-22 by Bombastisch