HiddenRamblings/TagMo #212: PowerTags NFC Commands

rowellx68, 2020-07-24:
I know that TagMo's support is considered to be alpha. However, I would like to get some clarifications to the NFC commands TagMo sends.

I'm not quite sure as to what `COMP_WRITE_CMD` does and why it precedes sending the modified `ptagKey`.

AbandonedCart, 2021-09-11:
If I understand correctly, it is a command to tell the hardware you are going to write. The N2 and PowerTags hardware are slightly different than a conventional tag and require "notice" of what is going to happen.

HiddenRambler, 2021-09-13:
This is such an old issue not sure if it's important any more but I'll answer anyway.

The ptagKey is essentially the "password" to make the tag rewritable again. This password has nothing to do with the conventional Amiibo passwords, but instead is a password unique to PowerTags. It's exact algorithm is unknown, but it is derived by some of the bits of the tag id. Currently I have included a dictionary of pre-generated passwords for some tag ids which can be used to unlock the tags with the same ids. This is also the reason that the app only uses set number of tag id's when programming PowerTags.

closed 2021-09-13 by HiddenRambler