HiddenRamblings/TagMo #215: Error while writing main data (Stage 1) android.nfc. TagLostException: Tag was lost

Nojiko444, 2020-09-05:
Hi there,
I'm having tons of problems with the app.
I spend hours already looking for solutions for everything and it seems that I was able to fix most of them.
But now I get the error "Error while writing main data (Stage 1) android.nfc. TagLostException: Tag was lost", as soon as I try to write the tag files on the NFC chip. 
And I have no idea anymore how to fix this. 
I already tested, if the NFC chips are right. And I got NTAG215 cards.
Also I have a Samsung Galaxy S9.

Bombastisch, 2020-12-22:
Take your phone case off, if you have one on your phone.

AbandonedCart, 2021-09-11:
It does not appear that any further reports were made, so I am assuming this was resolved. Please let me know if that was not the "case" (get it?)

closed 2021-09-11 by AbandonedCart