HiddenRamblings/TagMo #230: Large amounts of tags crash the application

witcheslive, 2021-03-29:
I downloaded a large archive of tags that would cause the application to crash even upon launch after using the Amiibo File Browser that necessitated me deleting my data and starting over. There is no error message, the app just quits back to the home screen.

I am able to use the app if I disable the Amiibo File Browser, but it would be nice to be able to use that with a large collection!

I am using a Samsung Note 20.

kaivar, 2021-06-10:
I have the same issue. It doesn't crash, but rather just goes back to the main screen instead of home screen. Pixel 3a on 11.

kaivar, 2021-06-10:
> I have seen the same issue intermittently in the past, but it was always
> happening with a particular set of downloaded bin files. I noticed that
> after I cleaned up the folder names of a downloaded archive to remove
> special characters (mostly "!"), it worked normally.
>
> It might be worth a shot.

Yep, that was it. Awesome, thank you!

closed 2021-09-11 by AbandonedCart