HiddenRamblings/TagMo #232: Transceive Failed, Tag was lost

WolfWyrm, 2021-04-08:
New to Tagmo, this is my first attempt trying to write Animal Crossing amiibos. I'm currently using [these cards from Amazon](https://www.amazon.ca/gp/product/B08P2C92WD/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1). I'm using a Google Pixel4a, each write attempt gives me java.io.IOExeption: transeive failed or Tag was lost HOWEVER, when I then scan the tag with Tagmo, it correctly detects the amiibo, but in-game, it fails to read the cards and gives me "That's... not an amiibo. Please use an Animal Crossing character's amiibo."

I've turned off tag validation with no change. I can attach logcat if need be.

AbandonedCart, 2021-09-11:
Is this still an issue with the newest experimental build?

WolfWyrm, 2021-09-11:
Unfortunately I don't have any tags left to test or I'd gladly give it a shot!

closed 2021-09-11 by AbandonedCart

AbandonedCart, 2021-09-11:
Closing this for now, but it can be opened again if you come across the same issue.