HiddenRamblings/TagMo #241: "Tag already an amiibo"

Dari-Once, 2021-07-04:
When I restore a tag (Marina from Splatoon 2) and says '_done_', when I scan that restored tag the app says that is the same amiibo as before the restore (Marina). 
And when I restore, load another tag to write it on a restored tag, also says that that tag had an amiibo.
Please help me, my phone model is a **Huawei P smart** and has **no phone case**

https://user-images.githubusercontent.com/86915936/124384866-915e0400-dcd3-11eb-80b0-3284053797fe.mp4

Artwebb1986, 2021-07-12:
I thought tags used for amiibo were write once ?

burndive, 2021-07-12:
Once an amiibo is assigned to a specific figure, you can't change what that figure is.  You can, however, update the data that is written to that amiibo (like the owner Mii and game data).

If you want to do the latter, you would tap "Restore" rather than "Write".  You can only "Write" once.  After that, you can "Restore" different versions of the same amiibo's data.

You use "Write" on a generic NTAG 215 to turn it into an amiibo (and also write the game data, if any).

You use "Restore" on an amiibo to give it game data captured from another amiibo (or the same amiibo at a different time).

Hope that helps.

AbandonedCart, 2021-09-11:
Closing this, as it appears to have an answer that "solves" it. Please feel free to comment if this was still an issue

closed 2021-09-11 by AbandonedCart