HiddenRamblings/TagMo #246: Sync with amiiboAPI still failing on Android 4.1.2

Buzbee, 2021-08-09:
I uninstalled previous version which had most of the icons. I am able to manually update with the amiibo.json but syncing still does not work. I lost all my images now too. Where are the images stored? Maybe i can restore them too if I can not sync.

AbandonedCart, 2021-08-09:
Have you tried running logcat when performing the sync? This might tell you what is happening.

Buzbee, 2021-08-09:
I downloaded logcat viewer. I am thinking maybe because this phone does not have a mobile connection (i am using wi-fi) that it is failing.
[image]

Buzbee, 2021-08-13:
I also noticed a possible oversight when changing from http to https://www.amiiboapi.com/api/amiibo/  in the source code for 2.7.0.

It is still using HttpURLConnection and not HttpsURLConnection to create a secure connection. This may solve my issue on android 4.1.2 . Looking at some code examples it doesn't look like you can just add the "s" either.

https://www.tabnine.com/code/java/classes/javax.net.ssl.HttpsURLConnection

AbandonedCart, 2021-08-13:
Your issue is with getting a response code, not the secure connection. 

According to your logs, the error is SettingsFragement:557 - https://github.com/HiddenRamblings/TagMo/blob/master2/app/src/main/java/com/hiddenramblings/tagmo/SettingsFragment.java#L557

Buzbee, 2021-08-13:
> Your issue is with getting a response code, not the secure connection.
> 
> According to your logs, the error is SettingsFragement:557 - https://github.com/HiddenRamblings/TagMo/blob/master2/app/src/main/java/com/hiddenramblings/tagmo/SettingsFragment.java#L557

Could this be a solution to my problem?

https://blog.dev-area.net/2015/08/13/android-4-1-enable-tls-1-1-and-tls-1-2/

AbandonedCart, 2021-08-13:
Yes, since it is identical to one of the solutions provided in the StackOverflow link.

Buzbee, 2021-08-13:
> Yes, since it is identical to one of the solutions provided in the StackOverflow link.

Can you implement it in your next version, TagMo-2.7.2-U.apk and reduce minsdk to 16 and I will test.

AbandonedCart, 2021-08-13:
The U builds are custom versions based on the PR for modernizing and are built to test the code. The purpose is to add support for newer devices, which would be the exact opposite of your request.

Buzbee, 2021-08-14:
> The U builds are custom versions based on the PR for modernizing and are built to test the code. The purpose is to add support for newer devices, which would be the exact opposite of your request.
> 
> The application does not rely on Google Play Services, so the issue with the "simple" implementation would be that now you are requiring every user to have Google Play Services because a handful have older devices.

How about a special build just for me then :0

Buzbee, 2021-08-14:
> How about a special build just for me then :0

I saw the Experimental restore of legacy device compatibility. Thank you. Let me know when is ready to test for everyone.

AbandonedCart, 2021-08-14:
Please do not confuse my PR with the original project. I am not a maintainer or contributor.

While I have submitted PRs to update the project, all aspects of the project outside the PRs are still part of the original project and should be treated as though they are intended for the original team.

GuiSalv10, 2021-08-14:
Please opt me out of this chat.

AbandonedCart, 2021-08-14:
@GuiSalv10 Visit the thread in your browser and click the unsubscribe button. Nobody tagged you, so you are receiving notifications because you opted into them.

Buzbee, 2021-08-15:
> While I have submitted PRs to update the project, all aspects of the project outside the PRs are still part of the original project and should be addressed to the original team.

I thank all the original team and those who have continued their work. I tried your latest 2.7.1 today with the Experimental restore of legacy device compatibility and your workaround works. Thank You. Unfortunately, the new version can not find my amiibos on my sd card. In 2.7.0 it would look in the folder and all subfolders in the directory I loaded the keys from. This is no longer happening. Even after moving them to the tagmo directory on the sd card they still are not found with that option checked in settings. I also noticed under Amiibo Info stats that the the 769 Amiibo are all showing as ERROR. The other 4 stats in settings are working fine.

AbandonedCart, 2021-08-15:
The custom builds attempt to use an external sdcard and fall back to internal storage. This is regardless of where the keys are loaded. 

It sounds like https://github.com/HiddenRamblings/TagMo/pull/247 will resolve this issue, though. I would assume the other issues are from the updates for newer Android.

Buzbee, 2021-08-15:
> The custom builds attempt to use an external sdcard and fall back to internal storage. This is regardless of where the keys are loaded.
> 
> It sounds like #247 will resolve this issue, though. I would assume the other issues are from the updates for newer Android.

The keys were loaded from the Amiibo dir originally on the sd card and i moved them to the the tagmo diectory with my amiibos on the sd card and reloaded the keys. I also tried copying the amiibos to internal storage in the tagmo dir. It still cant find any amiibos. I can sync with amiiboapi.com now because #247 did solve that issue but I can only scan tags though now. When you tap on Amiibos under Amiibos Info Stats under settings on your newer android device it does not display a bunch of ERRORs? That info comes from the amiibo.json file right? This happens before and after updating from amiiboapi.com for me.

AbandonedCart, 2021-08-15:
https://github.com/HiddenRamblings/TagMo/blob/master/app/src/main/java/com/hiddenramblings/tagmo/SettingsFragment.java#L182

This appears to be the location of the error, so the issue is in the original code.

Buzbee, 2021-08-15:
> https://github.com/HiddenRamblings/TagMo/blob/master/app/src/main/java/com/hiddenramblings/tagmo/SettingsFragment.java#L182
> 
> This appears to be the location of the error, so the issue is in the original code.

I took screenshots of the 2 issues comparing 2.7.0 to 2.7.1-U

[image]
[image]
[image]
[image]

The built in file explorer on 2.7.0 can display the folders on my sd card and thus find my amiibos in the first screenshot but not in 2.7.1 in the second screenshot. I am able to browse my sd card though in settings in 2.7.1 to select the location of the keys.

The original code works fine for display of the amiibo info on 2.7.0 in the third screenshot. ERROR on 2.7.1 as mentioned earlier in the fourth screenshot.

AbandonedCart, 2021-08-15:
Again, 2.7.1 was a custom build to test code. It is not made by the authors of this project. You should not be spamming their issues about it. I will not be responding any further.

Buzbee, 2021-08-15:
> https://github.com/HiddenRamblings/TagMo/blob/master/app/src/main/java/com/hiddenramblings/tagmo/SettingsFragment.java#L182
> 
> This appears to be the location of the error, so the issue is in the original code.

I am sorry. I thought I was helping you out and the rest of the team with testing your builds because they coincide with pull requests. I knew no other way to contact you about your builds. I am still willing to help if you have another communication method you prefer.

Although I do not have NFC on my android 10 device i tested your latest build on it. Both issues above exist on it also. I can get my amiibos on my sd card to show on Android 10 by clicking on the 3 dots and going to content and then unclicking Recursive Folders and then clicking it back. Alternatively, clicking on Show Missing Bins also works but crashes the app. Reopening the app also crashes it. I have to uninstall and reinstall it.

In the picture above on 4.1.2 the 3 dots are missing. This does indeed seem like its an issue from the updates for newer Android which is beyond the scope of your build for fixing I understand.

Good Luck to you and the rest of the team.

AbandonedCart, 2021-08-15:
Please stop abusing this issue as a means to contact me about a separate PR. I am NOT a maintainer of this project.

closed 2021-08-16 by Buzbee